fastmachinelearning/hls4ml #15: Interface for data type selection/tuning?

ejk43, 2017-11-11:
How would a user specify data types for the generated network?

A few possible scenarios come to mind:

- The converter uses a default width that is "okay" but could be improved (for example: 32 bits wide, which I believe is the current default)
- Bit widths are specified in a configuration file (float, 32 bits, 18 bits, 4 bits, whatever)
- Developer tunes bit widths manually after initially validating performance with larger types
- The python converter evaluates the network and asserts the best selection of integer/fractional bit widths

Or some combination of the above. Any other ideas?

nhanvtran, 2017-11-11:
Ah okay, was wondering about this in my comments to PR #14 but let's discuss here.  So in my mind I had:

"The converter uses a default width that is "okay" but could be improved (for example: 32 bits wide, which I believe is the current default)" 
(with a small modification that the default width could be in the YAML, like the default roll factor) 

with a little of 

"Developer tunes bit widths manually after initially validating performance with larger types"

I'm a little intrigued by door number 4 too (automated tuning).  

___

To keep you in the loop here, we were thinking once we have a full chain up and running (training repo + some validation plots), to loop in Song Han (who's going to be with Phil at MIT next fall).  We spoke with him about a month ago and he sounded like he was interested in giving us some advice how to best optimize networks on FPGAs and what are the most important hyperparameters.  

To that end, this reminds me I wanted to make an issue for network compression/pruning.  We thought about doing this on the training side, but there has to be some thought also put into the HLS translation as well.  Making issue number 17!

nhanvtran, 2018-02-19:
Our mode of operation is use defines the precision in the yml config and then can do finer tuned precision at the parameters.h level ...
closing unless there are burning issues

closed 2018-02-19 by nhanvtran