fastmachinelearning/hls4ml #16: Implement Resource-Reuse API for Fully-Connected Layer

ejk43, 2017-11-11:
I'd like to take a stab at ironing out what I'll tentatively call the "resource reuse API"-- ie, how the user of the compute_layer function will manipulate the resource usage of the core...

To that end: Can we come up with a few use cases we'd like to capture in this API? (the scope of the resource-usage problem is large enough that I'd prefer to start by agreeing on a few common use cases and design the function to cover these scenarios)

A few suggestions:

1. Totally unrolled, fully parallel layer
    - This is your general use case-- all features are consumed and operated on in parallel
    - Usage: Set fully_parallel option to true in the struct
2. Partially unrolled with 1-4 cycles of Initiation Interval (II)
    - Also useful for your application. Notionally, for an II=4, this should cut the multipliers by 4x and consume 1/4 of the features per clock cycle.
   - Usage: Set fully_parallel to false. Set target_initiation_interval to 4? Or do we want to use the "roll_factor"?
3. "Serial" operation with more unrolling
   - I'd classify this scenario as any situation where data is consumed serially (even if this is say 64-128 bits per clock cycle, which is plausibly common for lots of applications due to DMAs and other serial data transfer) 
    - Usage: Set fully_parallel to false. Maybe specify this operation by how many features are consumed per input cycle?

Does this all make sense so far? What am I missing?

ejk43, 2017-11-11:
An interesting data point for item 1... I chased down a few compiler directives needed to hit the II=1 with full parallel inputs. These were:

1. Outer-function PIPELINE directive in the computer layer
2. Outer-function PIPELINE directive in the activation layers
3. DATAFLOW directive in the top level C++ function (that contains multiple layers)

For some reason, I could NOT get the target II with UNROLL/PIPELINE directives *inside* the for-loops-- they needed to be outside the for-loops in each function. The trick may be to call this outer PIPELINE directive when we specify "full_parallel" operation

The relevant commit is in my branch here: https://github.com/hls-fpga-machine-learning/hls-fpga-machine-learning/pull/14/commits/0f66a886e198807a1c2f1ca9869858d850c303c5

nhanvtran, 2017-11-11:
The 3 use-cases you laid out are perfect.  Ideally we want fully unrolled but to balance latency with resource usage, Case (2) will be useful.  

Case (3) would be the slower "milliseconds" case but could be a good use-case for these CPU-FPGA accelerator hardwares like Brainwave and AWS F1.  (Also something we are ruminating on!)

Conceptual question: 
What's the difference between PIPELINE II=4 and not having anything pipelining and just using a roll factor of 4 for the cyclic array partition and the loop (un)roll?  

When we were fulling unrolling, we noticed it was the same resource usage and latency using pipelines or unrolls.  Also this is why we commented out the PIPELINE in the top function.  Better to do inside the layers and activations

ejk43, 2017-11-12:
> What's the difference between PIPELINE II=4 and not having anything pipelining and just using a roll factor of 4 for the cyclic array partition and the loop (un)roll?

Conceptually, the easiest case to think of is a single for-loop... Pipeline indicates that the loop needs to be able to accept a new input at a certain interval; while Unroll indicates that multiple iterations of the for-loop occur *at the same time*. In that sense, pipeline and unroll are not in fact mutually exclusive! A Pipeline II of 4 just means the loop needs to accept a new input on every fourth clock cycle-- it does not make any hard declarations about resource reuse. Partial unrolling DOES make hard declarations about resource usage, for example a partial unroll with factor 10 will generate 10 copies of the required for-loop logic (which then decreases the overall latency of the for-loop by a factor of 10). Unroll and Pipeline directives can also be "stacked" to both duplicate the for-loop logic and to force a lower II. 

In general, I've found the PIPELINE II=X option to be an unreliable way to affect compiler behavior... So, to answer your question, the partial unroll with array partition would be more effective at defining resource usage than a PIPELINE II=4.

Also, I definitely agree the best option is to pipeline as "far inside" the functions as possible. If pipeline is declared at the outer level, all internal directives for partial unrolling are ignored-- an unfortunate side effect of the PIPELINE directive.

ejk43, 2017-11-12:
> In general, I've found the PIPELINE II=X option to be an unreliable way to affect compiler behavior... So, to answer your question, the partial unroll with array partition would be more effective at defining resource usage than a PIPELINE II=4.

... Interestingly enough, after playing around with Scenario 2 a little more, I'm very puzzled how to make the compiler happy. In fact the best approach I've found so far is to pipeline the full "compute_layer" function and hope that logic improves with a higher II (it seems to be questionable whether PIPELINE II =4 is really any better than normal pipeline with specified II). 

I think the core problem here stems from the cascaded adder logic in the matrix multiply, plus the way that HLS wants to pass data between functions. I've had a lot more luck so far with the "serial" approach, but the full parallel approach may be best handled with the aggressive pipeline directives. Not totally sure yet-- will keep looking at it...

nhanvtran, 2017-11-12:
Thanks for the explanation, trying to wrap my head around it. 

I played around with it a little this morning and I see what you mean about puzzling performance.  Ideally we would have fully unroll take up resources X and total latency L.  Then when we have a roll factor of 4, it would take resources X/4 and total latency 4*L.  

However, with a roll factor of 4, I managed to get a factor of 8*L latency and resources 4*X! 😮 

And whenever I add a PIPELINE in the top function, everything just gets fully flattened.  

(I'm working exclusively with the 1-Layer example if that helps with a common discussion)

nhanvtran, 2017-11-12:
Playing a little more with a very simple loop
See branch `nvt/testing-branch`

So I managed to get the reduced LUT resources and increased latency with partial unroll (though DSP and FFs went up).  It requires unrolling with the right factor and a corresponding cyclic partition of the array.  But whenever I add a PIPELINE, it ruins everything!

*Partition Complete, Unroll 12:*
|  Latency  |  Interval | Pipeline |
|    4|    4|    5|    5|   none  |
|       Name      | BRAM_18K| DSP48E|   FF   |   LUT  |
|Total            |        0|      2|       5|     193|

*Partition Complete, Unroll 6:*
|  Latency  |  Interval | Pipeline |
|   51|   51|   52|   52|   none  |
|       Name      | BRAM_18K| DSP48E|   FF   |   LUT  |
|Total            |        0|     12|      35|     246|

*Partition Cyclic 6, Unroll 6:*
|  Latency  |  Interval | Pipeline |
|   13|   13|   14|   14|   none  |
|       Name      | BRAM_18K| DSP48E|   FF   |   LUT  |
|Total            |        0|     12|      44|      44|

*Partition Cyclic 6, Unroll 6, Pipeline 2:*
|  Latency  |  Interval | Pipeline |
|    5|    5|    2|    2| function |
|       Name      | BRAM_18K| DSP48E|   FF   |   LUT  |
|Total            |        0|      2|     325|     261|

nhanvtran, 2017-11-13:
Hm.....
https://forums.xilinx.com/t5/Vivado-High-Level-Synthesis-HLS/How-can-I-prevent-pipelined-function-from-unroll/td-p/719898

ejk43, 2017-11-13:
Hah, yes I believe that post is correct. That's generally why I prefer dataflow over pipeline, but pipeline does appear to be quite effective for our parallel input case. 

How many DSPs would be required in your example for a full unroll?

I've got a new idea to try: explicitly split the multipliers and the accumulators into different for-loops. We should be able to partially until the multipliers to get a dsp reduction, then pipeline the accumulators.

nhanvtran, 2017-11-13:
@ejk43 

@benjaminkreis @jmduarte and I played around a little more today and had some moderate success with the pipelining with a simple example loop (see code below).  This allowed us to halve the DSP resource usage and double the latency effectively (it's all in this `rewind` option).  

However, with the nested loops in `compute_layer` getting the pipelines of the different loops to match up is a little tricker.  So we thought about flattening the 2D loop into one long 1D loop -- what do you think?  

We could also then split out the multipliers and accumulators as well.  Just thinking out loud here

______________________________________________________________________________

```
void myproject(
		  ap_fixed<32,16> data[12],
		  ap_fixed<32,16> res[12])
{
    #pragma HLS ARRAY_PARTITION variable=data cyclic factor=6 dim=1
    #pragma HLS ARRAY_PARTITION variable=res cyclic factor=6 dim=1    
    for(int iacc = 0; iacc < 12; iacc++) {
        #pragma HLS PIPELINE II=1 rewind
        #pragma HLS UNROLL factor=6
        res[iacc] = data[iacc]*123;
    }
}
```

ejk43, 2017-11-14:
Yes! This is a great idea. I've been struggling to get the unrolled for-loop to restart the next "unrolled" iteration before waiting for the entire loop latency, but I forgot about the rewind option! This actually let me hit a good initiation-interval for the matrix-multiply:

```C++
for(int ii = 0; ii < CONFIG_T::n_in; ii++) {
    if (!CONFIG_T::full_parallel){
        #pragma HLS UNROLL factor=unroll_factor_in
        #pragma HLS PIPELINE rewind
    }
    for(int jj = 0; jj < CONFIG_T::n_out; jj++) {
        mult[ii][jj] = cache[ii] * weights[ii][jj];
    }
}
```

You're definitely right about 2D loops being an issue with partitioning, since the matrix multiply needs to access multiple columns (or rows) at each clock cycle, which doesnt lend itself well to the array partition operation. Flattening to 1D could be an option. For parallel inputs, perhaps a reasonable approach is just complete partitioning-- it takes more logic, but we definitely need to read a good amount of data at each clock cycle in parallel, which BRAM is not optimized for.

But here's the next problem: When I split up the multiply and accumulate operations to optimize the multiplier reuse, I needed to use a function-level DATAFLOW operation. I *believe* the DATAFLOW operation is implementing a ping-pong buffer behind the scenes, which is overkill for this scenario. It instantiates far too much logic (because it cannot used BRAM since there's so much parallel operations/ everything is currently array_partitioned "complete"). 

I ended up finding perhaps a bit of a hack solution, but it has definitely given me the best performance and reliability:

```C++
int multiplier_limit  = ceil(CONFIG_T::n_in*CONFIG_T::n_out / CONFIG_T::unroll_factor);
#pragma HLS PIPELINE
#pragma HLS ALLOCATION instances=mul limit=multiplier_limit operation

Product1: for(int ii = 0; ii < CONFIG_T::n_in; ii++) {
    cache = data[ii];
    Product2: for(int jj = 0; jj < CONFIG_T::n_out; jj++) {
        mult[ii][jj] = cache * weights[ii][jj];
    }
}
accumulator<res_T, CONFIG_T>(mult, biases, res);
```

Note the HLS ALLOCATION pragma. I'll also add: strangely, there is a significant performance improvement between putting the accumulator in its own subfunction vs leaving it inline with the multiply. 

Here's a few comparisons of synthesis results from one of the examples (available on my ejk/resource-reuse-api-branch). For reference, the fully-pipelined version, with no resource reuse, requires 270ish DSPs and about 10k each FFs and LUTs (the total multipliers based on network size should be 32*11,  so I suspect HLS is already doing some amount of network compression):

1. Dataflow + Unroll/Pipeline/Rewind + accumulator in its own function
    - Interval: 4
    - BRAM: 1
    - DSP48: 88
    - FF: 9718
    - LUT: 11991
2. Pipeline + Multiplier allocation limit + accumulator in its own function:
    - Interval: 4
    - BRAM: 1
    - DSP48: 88
    - FF: 10.7k
    - LUT: 18.5k
3. Pipeline + Multiplier allocation limit + multiply and accumulate in same loop
    - Interval: 4
    - BRAM: 1
    - DSP48: 88
    - FF: 9.8k
    - LUT: 12k

Very peculiar.. I'm happy to use this ALLOCATION approach, and keep the accumulator in a subfunction, but it really makes me wonder why that's necessary. (HLS... FML...)

ejk43, 2017-11-14:
So to revise my initial API suggestions, after playing around with options for a while, I have slightly updated my recommended approach... We should handle two possible use-cases, where each option includes some unrolling/resource reuse capability:

1. Parallel Inputs (all features consumed at once)
2. Serial Inputs (features consumed serially)

For Parallel Inputs, no resource reuse suggests the HLS function can consume a new feature set every clock cycle; while 2x resource reuse would consume 1/2 the multipliers but only consume a new feature every 2 clock cycles; and so on. For Serial Inputs, we would have largely the same paradigm, except with a single feature per epoch. 

I'm currently using a combination of flags in the API:
 - io_type: which accepts either nnet:io_parallel or nnet::io_serial
 - reuse_factor: Indicates how much resource reuse to use / how much to constrain multipliers. Empirically, I've seen that HLS actually does a pretty good job identifying the required II to support a target number of multipliers. How about that!

These changes are captured in an updated PR for the API changes, which will currently break the C++ generation python script as-is, but could be worth a look anyway. https://github.com/hls-fpga-machine-learning/hls-fpga-machine-learning/pull/18

Anything else missing for a first cut? There's already a variable in the struct API for using BRAMs, which is currently unimplemented-- I'd suggest maybe trying to finish a first pass of squeezing weights into BRAMs... Definitely, we'd reduce some logic requirements if we could convert some of the weights to BRAM, even though it may not be well suited for such a parallel operation, it would still be useful for more serial operations.

ejk43, 2017-11-14:
This was nagging me for a while: 

>I'll also add: strangely, there is a significant performance improvement between putting the accumulator in its own subfunction vs leaving it inline with the multiply.

When I went to try it again this morning with a slightly different architecture, I was not able to replicate the poor results. Perhaps something changed that resolved the compiler issue? Anyway, updated the branch to remove the subfunction: https://github.com/hls-fpga-machine-learning/hls-fpga-machine-learning/pull/18/commits/dc66e33dccc4bce3d6cdffbc80b44b47474dcad7

nhanvtran, 2018-02-19:
this is working now and our current branch uses the API, closing...

closed 2018-02-19 by nhanvtran