fastmachinelearning/hls4ml #21: Automated testing/continuous integration?

ejk43, 2017-12-04:
Just tossing an idea out here, but would there be any interest in setting up a small automated test framework to "protect" some of the intended behavior of the resource reuse API? 

I'm starting to get afraid that something I edit may accidentally break existing features. Thoughts? Does anyone have experience with Jenkins or Travis CI? I believe they both provide free access for open source projects.

nhanvtran, 2017-12-04:
I have no experience in this, but I would be interested in getting some automated testing.  We use Jenkins on CMS so I have a slight preference for that, but I honestly don't even know where to start.

vloncar, 2018-07-26:
Given the amount of bugs I have encountered over the last few days, I think we should do something about this and I would like to volunteer. The way I see it, there are few integration options, we can either go with cloud-based offering like TravisCI or a more traditional one like Jenkins.

TravisCI has nice integration with GitHub and is very popular on GitHub, but is kind of limited in the amount of parallelism we get for free as well as flexibility. OTOH, we can use Jenkins service provided by CERN. It will be a little bit more difficult to set up and connect with our github project, but is much more flexible.

The current trend in CI seems to be using Docker images to create the build and test environment. If we go that way (which I would much prefer), we should create an image with Vivado HLS. This raises the question of licencing, but since we won't be distributing the image and will keep it in a private repository, we should be ok.

I think we can start out small, just test the basic tools (`keras-to-hls` and `hls-writer`), then add Vivado HLS. I have a small preference for TravisCI (its shiny and new), and have used Jenkins a long time ago (for a Java project back when it was forked off from Hudson). However, if we create the Docker image and accompanying tests in it, setting up any CI around it should not be a huge challenge.

nhanvtran, 2018-07-26:
Hi @vloncar thanks for the interest in taking this on!  We've only had some preliminary discussions about what to do here, so there are no preferred solutions at the moment.  The main thing to worry about, which you've identified, is that we may run into Vivado HLS licensing issues.  

So in this case, maybe doing it via docker might be the best way to go.  Otherwise, I got a little scared by this line in the WIki article: "Although the source is technically free software and available piecemeal on GitHub under permissive licenses, the company notes that it is unlikely that casual users could successfully integrate it on their own platforms.[4]"
https://en.wikipedia.org/wiki/Travis_CI

That said, feel free to explore what is possible and we can go from there.

ejk43, 2018-07-27:
@vloncar: In other projects, I'm currently use gitlab's ci workflow for Vivado builds. It's super easy to set up, takes maybe 10 minutes if you already have a vivado docker image ready. The two big features that make it work are:
1) Create a self hosted test runner. I have a computer that lives in my lab running the "gitlab-runner" application which is registered with my gitlab project. This computer also has access to the floating license server.
2) Use a specified docker image environment. I have a docker image that basically contains Vivado and git. It's so much easier to build/run CI tests in a docker image since it's already inititialized and you dont need to reinstall Vivado every time.

The CI process then runs tests using the Vivado docker image environment on my own computer, which is connected to our Vivado license server, so it can pull correct licenses. 

Unfortunately, it looks like this self-hosted paradigm is *not* available on free open-source travis-ci. The website is not totally clear on this, but it looks like you need Enterprise version (https://enterprise.travis-ci.com/). For that reason, even though Travis CI uses the nice yml syntax, I'd probably recommend investigating Jenkins...

ejk43, 2018-08-24:
Hey @vloncar, any luck setting up a CI system for HLS synthesis? I'm getting nervous to edit things already and would love to have some amount of testing to make sure I dont break things everyone expects to work.

vloncar, 2018-08-24:
Hi @ejk43, I have created a set of shell scripts and a Jenkins pipeline, tested it, and am now ready to set it all up on correlator2 machine at Fermilab. You can see the WIP code in jenkins-ci branch on my repo, and you can use the scripts outside CI if you want to do some manual testing. I am away from my computer for a few more days, so I hope to have something set up by the end of next week.

nhanvtran, 2018-09-05:
just a small status here, I tested the PR that was merged from @vloncar with standalone scripts and it works great!  we are waiting for some web permissions from FNAL to enable jenkins

closed 2018-11-29 by nhanvtran