fastmachinelearning/hls4ml #27: Next version of CNN not hitting pipeline target

benjaminkreis, 2018-01-03:
Hi folks,

I made some big changes to the CNN code so that it:
1. - Allows for multiple filter
2. - Sums over channels within a filter
3. - Configurable stride
4. - Configurable padding

The csim results match within a few percent for the one example I've tried, but unfortunately HLS can't hit the interval=1 target and gets 2 instead.  I haven't yet been able to figure out exactly what's causing it, but perhaps some relevant output is this:
`INFO: [SCHED 204-61] Pipelining function 'conv_1d.0.0.0.0'.
WARNING: [SCHED 204-69] Unable to schedule 'store' operation (/home/kreis/conv/HLS4ML/nnet_utils/nnet_conv.h:133) of variable 'acc[0][2].V', /home/kreis/conv/HLS4ML/nnet_utils/nnet_conv.h:122 on array 'res_0_V' due to limited memory ports.
INFO: [SCHED 204-61] Pipelining result: Target II: 1, Final II: 2, Depth: 6.
WARNING: [SCHED 204-21] Estimated clock period (4.89ns) exceeds the target (target clock period: 5ns, clock uncertainty: 0.625ns, effective delay budget: 4.38ns).
WARNING: [SCHED 204-21] The critical path consists of the following:
	'mul' operation ('p_Val2_257_2', /home/kreis/conv/HLS4ML/nnet_utils/nnet_conv.h:98) (4.89 ns)`

I tried paring down the computations (lower number of multiplications and precision, change multiplications to additions, less accumulation etc) and could only get to interval=1 in some pretty specific circumstances that I don't think provide any great insight.  

I'm still investigating, but I thought I'd create this issue in case anyone already has some ideas.

The main branch I'm developing on is [this one](https://github.com/hls-fpga-machine-learning/HLS4ML/blob/bjk/conv1d/nnet_utils/nnet_conv.h)

And I have a second one where I tried separating the accumulator loop within the filter and over channels into two [here](https://github.com/hls-fpga-machine-learning/HLS4ML/blob/bjk/conv1d_sep/nnet_utils/nnet_conv.h)

The [example project](https://github.com/hls-fpga-machine-learning/HLS4ML/tree/bjk/conv1d/example-prjs/conv-1layer) is updated with a 1D CNN I trained in Keras.

benjaminkreis, 2018-01-03:
One more observation -- I can hit the target interval of 1 if I use this code (no simplifications/paring down of the computations) but make the network much smaller (e.g. 16 inputs, valid padding, 2 channels, 2 filters, filter width 2, stride 3).  This only uses 30 DSPs.  (And I still had to increase the clock period to get rid of the warning about that.)

benjaminkreis, 2018-01-10:
So in fact, just generally for smaller CNNs, things are looking alright.  So I'm closing this issue for now and will make a PR for the code.  We will rediscover missing the pipeline target as explore CNN size, but I don't think we need to consider this an open issue at the moment.

closed 2018-01-10 by benjaminkreis

ejk43, 2018-01-11:
This is curious behavior... I would expect HLS should pipeline the code such that you can hit a target II successfully (especially with the HLS PIPELINE directive) . I've hit a related issue if I'm also imposing a latency constraint, but I dont think there's a latency constraint here? 

Seems a little odd that the value that worked for you was a filter width of 2 -- seems very small to me. Do you have any feel for exactly where the HLS synthesis starts to miss your targets?

Also, this warning you got is a little strange, too: `WARNING: [SCHED 204-69] Unable to schedule 'store' operation (/home/kreis/conv/HLS4ML/nnet_utils/nnet_conv.h:133) of variable 'acc[0][2].V'`

The warning about a "store" operation is typically associated with trying to read/write too much info into a BRAM. Is acc fully partitioned? If fully partitioned, HLS should implement acc in fabric, rather than BRAM, which I would THINK should avoid the store operation warning.. ?

ejk43, 2018-01-11:
Another thought on this topic: I believe the HLS build scripts set the target clock period to 5 ns, which implies a 200 MHz clock rate. 

So if you're applying input data at 40 MHz, your application could run with initiation interval of 5 (rather than initiation interval of 1), which definitely relaxes some of the timing issues. Alternatively, you could set the HLS clock period to 10-20 ns instead of 5 ns, which would allow HLS to connect more fabric between clock cycles.. Might be good to make the clock period configurable, rather than hardcoded to 5 ns?

Of course, the key thing to watch out for is if HLS doesnt fully saturate your DSPs... We want all DSPs running at every clock cycle to get maximum fix-pt OPS

benjaminkreis, 2018-01-11:
Hi @ejk43, thanks for the feedback!

You are correct that I did not impose a latency constraint.  When I missed the pipeline target, I was trying bigger networks, for example with 32 inputs in 4 channels, with a few filters of width 5 or so.  But I haven't yet mapped this out enough to know when it starts to happen.  We will definitely be looking into that next.  

The acc and other arrays in the function did have the complete array partition, so indeed it is strange.  

I agree with what you said on the clock period and pipelining for 40 MHz.  At one point I changed it to period=20 and the timing error went away, but if I recall correctly, it still didn't hit the pipeline target.  It was kind of a mess with so many things going wrong with the bigger networks.  Hopefully we can disentangle everything by starting with a smaller one that works and increasing it!

I'm going to make an issue for putting the clock period in the yaml.

Thanks again!