fastmachinelearning/hls4ml #28: Precision too low causes synthesis crash

nhanvtran, 2018-01-03:
@zhenbinwu reports that running with a precision at ap_fixed <8,4> crashes 1 layer example need to figure out why

benjaminkreis, 2018-01-03:
Where does it crash?  I just tried, and it looked okay to me.

But this doesn't surprise me.  I also found that the cosim output didn't match the csim output for certain choices of precision.  I can't remember if it crashed then, though.

zhenbinwu, 2018-01-03:
`INFO: [XFORM 203-622] Instantiating function 'nnet::compute_layer<ap_fixed<8, 4, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<8, 4, (ap_q_mode)5, (ap_o_mode)3, 0>, config1>.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0'(/home/benwu/HLS4ML/nnet_utils/nnet_layer.h:61:28) to 'nnet::compute_layer<ap_fixed<8, 4, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<8, 4, (ap_q_mode)5, (ap_o_mode)3, 0>, config1>.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0' at call site (firmware/myproject.cpp:58) by setting 'biases[31].V' to 'b1.V.31'.
Global variable initializer type does not match global variable type!
[1 x i8]* @w2.V.31
Global variable initializer type does not match global variable type!
[1 x i8]* @w2.V.2
Global variable initializer type does not match global variable type!
[1 x i8]* @w2.V.17
Broken module found, compilation aborted!
Stack dump:
0.      Running pass 'Function Pass Manager' on module '/home/benwu/HLS4ML/keras-to-hls/test_DefaultPrecision8a4_ReuseFactor6/myproject_prj/solution1/.autopilot/db/a.o.1.bc'.
/home/xilinx/Vivado_HLS/2016.4/bin/loader: line 179:  9910 Aborted                 "$RDI_PROG" "$@"
`

It actually crash on csim step. It doesn't work for ap_fixed<4, 4> and ap_fixed<8, 4>.

zhenbinwu, 2018-01-03:
Running standalone can't reproduce this error. Double checking..

zhenbinwu, 2018-01-03:
crashes with ap_fixed<4, 4> and ap_fixed<8, 4> appeared in code, which latest commit dated Dec. 22 2017. Update to the most recent code seems fix this issue. Closing this issue

closed 2018-01-03 by zhenbinwu

nhanvtran, 2018-01-04:
what was the resolution on this?

nhanvtran, 2018-01-04:
sorry... missed the last message, ok seems all good now