fastmachinelearning/hls4ml #36: Use ARRAY_RESHAPE directives at the top level myproject.cpp?

ejk43, 2018-01-27:
Hey guys, 

Just ran into something interesting about the top-level interfaces... By default it seems that the ARRAY_PARTITION directive instantiates a separate port for each partitioned element. So, if you have an array with N elements, you'll have N separate data ports, which would get rather large for the number of inputs you'll probably need. 

I *think* if we replace the ARRAY_PARTITION directive with ARRAY_RESHAPE, we get a similar impact as partitioning the array-- and the array elements will also be concatenated into a single larger array element. Here's my updated directives: 

```C++
    #pragma HLS ARRAY_RESHAPE variable=data complete dim=0
    #pragma HLS ARRAY_RESHAPE variable=res complete dim=0
    #pragma HLS INTERFACE ap_hs port=data,res
```

Here's the port output for Nhan's conv1d example: (notice data port is 768 bits wide, result port is 192 bits wide)

```
* Summary: 
+-----------------------+-----+-----+------------+----------------+--------------+
|       RTL Ports       | Dir | Bits|  Protocol  |  Source Object |    C Type    |
+-----------------------+-----+-----+------------+----------------+--------------+
|ap_clk                 |  in |    1| ap_ctrl_hs |    myproject   | return value |
|ap_rst                 |  in |    1| ap_ctrl_hs |    myproject   | return value |
|ap_start               |  in |    1| ap_ctrl_hs |    myproject   | return value |
|ap_done                | out |    1| ap_ctrl_hs |    myproject   | return value |
|ap_idle                | out |    1| ap_ctrl_hs |    myproject   | return value |
|ap_ready               | out |    1| ap_ctrl_hs |    myproject   | return value |
|data_V_ap_vld          |  in |    1|    ap_hs   |     data_V     |    pointer   |
|data_V                 |  in |  768|    ap_hs   |     data_V     |    pointer   |
|data_V_ap_ack          | out |    1|    ap_hs   |     data_V     |    pointer   |
|res_V_ap_ack           |  in |    1|    ap_hs   |      res_V     |    pointer   |
|res_V                  | out |  192|    ap_hs   |      res_V     |    pointer   |
|res_V_ap_vld           | out |    1|    ap_hs   |      res_V     |    pointer   |
|const_size_in          | out |   16|   ap_vld   |  const_size_in |    pointer   |
|const_size_in_ap_vld   | out |    1|   ap_vld   |  const_size_in |    pointer   |
|const_size_out         | out |   16|   ap_vld   | const_size_out |    pointer   |
|const_size_out_ap_vld  | out |    1|   ap_vld   | const_size_out |    pointer   |
+-----------------------+-----+-----+------------+----------------+--------------+
```

I'd be interested if this breaks any of the other examples, but I'd suggest it's probably a good idea to swap over for the top level function (maybe also for the subfunctions too??)

I threw a test together based on Nhan's conv1d branch here: https://github.com/hls-fpga-machine-learning/HLS4ML/tree/ejk/conv1d-array-reshape

nhanvtran, 2018-01-28:
Cool!  I think in terms of book-keeping with wiring up different IPs this makes a lot of sense.

benjaminkreis, 2018-01-29:
Cool indeed!  Is this only suggested for the top level ports?

benjaminkreis, 2018-01-29:
A few things we should do:
- Change to `ARRAY_RESHAPE` for data and res in dense layer, too
- Add dim=0 to reshape/partition inside nnet_layer.h (e.g. it's missing from `mult`)
- Test for curiosity's sake if `dim=0` or `ARRAY_RESHAPE` allow us to put this back in for `weights` which was crashing with some compressed models

closed 2019-04-04 by benjaminkreis