fastmachinelearning/hls4ml #38: Clarify which keras layers we support in documentation

jmduarte, 2018-01-31:
We should list which ones we currently support and which ones we plan to support in the documentation.

Below is a list of all Keras layers, not including abstract layers and aliases.

```
Activation
ActivityRegularization
Add
AlphaDropout
AtrousConv1D
AtrousConv2D
Average
AveragePooling1D
AveragePooling2D
AveragePooling3D
BatchNormalization
Bidirectional
Concatenate
Conv1D
Conv2D
Conv2DTranspose
Conv3D
Conv3DTranspose
ConvLSTM2D
ConvRecurrent2D
Cropping1D
Cropping2D
Cropping3D
CuDNNGRU
CuDNNLSTM
Dense
Dot
Dropout
ELU
Embedding
Flatten
GRU
GRUCell
GaussianDropout
GaussianNoise
GlobalAveragePooling1D
GlobalAveragePooling2D
GlobalAveragePooling3D
GlobalMaxPooling1D
GlobalMaxPooling2D
GlobalMaxPooling3D
Highway
Input
LSTM
LSTMCell
Lambda
LeakyReLU
LocallyConnected1D
LocallyConnected2D
Masking
MaxPooling1D
MaxPooling2D
MaxPooling3D
Maximum
MaxoutDense
Merge
Minimum
Multiply
PReLU
Permute
RepeatVector
Reshape
SeparableConv1D
SeparableConv2D
SimpleRNN
SimpleRNNCell
Softmax
SpatialDropout1D
SpatialDropout2D
SpatialDropout3D
StackedRNNCells
Subtract
ThresholdedReLU
TimeDistributed
UpSampling1D
UpSampling2D
UpSampling3D
ZeroPadding1D
ZeroPadding2D
ZeroPadding3D
```

Duchstf, 2019-08-14:
@jmduarte Is this the full list of keras layers we plan to support in the future? Currently I think we have less than that?

ProjectMarlin, 2022-05-18:
Hi, is there a version of hls4ml that does support LSTM layers? Currently I am having trouble converting my small LSTM model to hls implementation using hls4ml v0.6.0 in python. Thank you.

spacenets7, 2026-11-02:
Is there still interest in extending this beyond the architecture-level support table? I’d like to propose a small layer-support matrix recording frontend, backend, IO mode, and important restrictions, with entries backed by existing tests where possible. Would a limited first slice covering recurrent layers be useful?