fastmachinelearning/hls4ml #49: Understanding HLS multipliers

benjaminkreis, 2018-02-20:
@ejk43 I was hoping to do some simple tests with a single multiplication in HLS to get a more quantitative understanding of the latency, II, and DSP usage.  We know the trends qualitatively, such as more DSPs and latency for higher precision, but I'd love to know the numbers.

HLS must have some freedom in determining these things.  First of all, when you create a multiplier IP in normal Vivado, you have choices for the number of pipeline stages, a.k.a. latency, even for a fixed precision.  

I also see this in some basic tests:

In a trivial HLS project with one multiplication of ap_fixed<18,8> numbers with PIPELINE=1, I got a multiplier with a latency of 1:
```
    test_mult_mul_mulbkb_U1 : component test_mult_mul_mulbkb
    generic map (
        ID => 1,
        NUM_STAGE => 1,
        din0_WIDTH => 18,
        din1_WIDTH => 18,
        dout_WIDTH => 28)
    port map (
        din0 => b_V,
        din1 => a_V,
        dout => p_Val2_s_fu_67_p2);
```

However in our 1 hidden layer example, also with ap_fixed<18,8>, I see that the number of stages is 3:
```
    myproject_mul_11nfYi_U7 : component myproject_mul_11nfYi
    generic map (
	ID => 1,
        NUM_STAGE => 3,
        din0_WIDTH => 11,
        din1_WIDTH => 18,
        dout_WIDTH => 28)
    port map (
        clk => ap_clk,
        reset => ap_rst,
        din0 => grp_fu_743_p0,
        din1 => grp_fu_743_p1,
        ce => grp_fu_743_ce,
        dout => grp_fu_743_p2);
```

So my question is, how does HLS make these choices?

Do you know if there are some basic rules for this, or does it really depend case-by-case on the whole project routing, in which case I have no hope of mapping this out with a single multiplier project?

ejk43, 2018-02-22:
First, I admit I'm still a little confused-- what's the end goal you're working towards here? Overall I'm a little skeptical about attempting to back out the quantitative rules behind HLS optimizatons by stimulating the input and observing output HDL. It does not seem like an easy or efficient way to quantitatively understand the operation of the HLS tool. While we could perhaps find the actual rules -- maybe codified or in documentation somewhere-- in general I believe the HLS concept is that the designer *doesnt* need to worry about that :P ... We dont try to "look under the hood", when converting from HDL to the bitstream logic mapping, right? (...or at least, I dont)

Anyway, about latency: I have not looked into the NUM_STAGE parameter before, but I would not be surprised if that has to do with hitting timing. There's some physical register stages built into Xilinx DSPs that can achieve optimal clock rates when the designer specifies HDL registers before/after the DSP core. Perhaps an interesting test to help you get more of a feel for behavior would be to set the desired timing to 20 ns or so (5 ns is pretty aggressive, but I like to design here because the same design would easily apply to lower clock rates). Anecdotally I've observed higher register usage, II, and latency when designing for tighter timing constraints -- I believe implementing a design to hit a particular timing target is definitely part of the "black magic" behind the HLS tool that could be tough to understand with a single multiplier project, but you might be able to see smaller NUM_STAGEs with a larger clock period.

Certainly, relatively easy parameters to observe here would be the din0_WIDTH and din1_WIDTH; these should be driven directly by the selected fixed point bit width and should be deterministic. Of course, this might still get complicated if there's partial weight optimizations, or if the larger fixed-point multiplications need to be spread across multiple DSPs.

Definitely a tricky question!

benjaminkreis, 2018-02-22:
Ha sorry I gave no context here, and I definitely don't want to reverse engineer all of HLS :).  I was hoping to add some simple advice to the hls4ml documentation about how reducing precision saves on resources and latency.  It's pretty straight forward for DSP resources because we know single DSP48s do up 18x25 bit multiplication.  I was hoping to provide an example for latency, too, such as "it takes X clocks to multiply two <18,8> numbers but Y clocks to multiply two <32,12> numbers".  But to my surprise when I made a project with a single multiplication, it didn't look like the neural net examples, ie i got latency=1.  The lesson is that there's a lot of freedom for optimizing the multipliers, so probably it's best to leave the quantitative statements on latency to plots like @jngadiub and @zhenbinwu are making.

closed 2018-02-22 by benjaminkreis