fastmachinelearning/hls4ml #56: Pruning conv1d

benjaminkreis, 2018-03-28:
I have a [branch](https://github.com/hls-fpga-machine-learning/hls4ml/tree/bjk/prune_conv) going to allow the conv1d to take advantage of pruning in training.  @jmduarte and I have been discussing this, and we think there isn't a simple formula for reducing the multiplier limit based on the number of weights equaling zero.  It gets pretty complicated due to the presence of padding.  

So the approach here is to just do the loops once in advance and count the number of multiplications by nonzero weights, then divide that by the reuse factor to get the multiplier limit.  

Perhaps the simplest would be to do these first loops for counting in separate code (e.g. our python), but I think it would be nice for the nnet_utils to have this feature.  So I've added the count [here](https://github.com/hls-fpga-machine-learning/hls4ml/blob/bjk/prune_conv/nnet_utils/nnet_conv.h#L108).  Unfortunately it looks to me like I haven't managed to fully decouple it from the firmware part.  I see a slightly different resource usage if I use this function versus just passing the limit from the outside world.  If anyone sees the problem, let me know!

Some test results:
Model: KERAS_conv1d_small
Precision: ap_fixed<18,8> 
HLS Version: v2017.2

Default weights:
Reuse | BRAM | DSP | FF | LUT | Lat | II
--- | --- | --- | --- | --- | --- | ---
1 | 13 | 547 | 47149 | 28161 | 24 | 1
2 | 13 | 398 | 47188 | 30769 | 26 | 2
3 | 13 | 266 | 47504 | 33031 | 30 | 3

Random 20% weights set to zero:
Reuse | BRAM | DSP | FF | LUT | Lat | II
--- | --- | --- | --- | --- | --- | ---
1 | 13 | 455 | 41276 | 23950 | 24 | 1
2 | 13 | 303 | 37926 | 24188 | 26 | 2
3 | 13 | 206 | 38074 | 26472 | 30 | 3

Random 50% weights set to zero:
Reuse | BRAM | DSP | FF | LUT | Lat | II
--- | --- | --- | --- | --- | --- | ---
1 | 13 | 279 | 24769 | 15877 | 22 | 1
2 | 13 | 200 | 27442 | 16876 | 25 | 2
3 | 13 | 99 | 17592 | 11563 | 25 | 3

I'm setting the mult limit, so I think here the test is to see if the non-DSP numbers look problematic.  I don't see the FF or LUT numbers explode suggesting we are doing multiplications with logic.  Latency looks alright.  II is as expected.

So if people like this approach, I can make a PR.  It would be nice to first figure out why the mult count is possibly taking some firmware to do though.

benjaminkreis, 2018-03-30:
I dug into the whether or not the new multiplier count is taking resources.

What I see is a bit weird, but I think we can move on.  I'll summarize it here, though.

If I compute the number of multiplies in the new function compute_n_mult then in conv_1d make the muliplier limit as so:
```
const int multiplier_limit = ceil( float(compute_n_mult<CONFIG_T>(weights)) / float(CONFIG_T::reuse_factor) );                                                                                                                                                            
#pragma HLS ALLOCATION instances=mul limit=multiplier_limit operation
```

I get the following resource usage:
```
+-----------------+---------+-------+--------+--------+
|       Name      | BRAM_18K| DSP48E|   FF   |   LUT  |
+-----------------+---------+-------+--------+--------+
|DSP              |        -|      -|       -|       -|
|Expression       |        -|      -|       0|       6|
|FIFO             |        -|      -|       -|       -|
|Instance         |       13|    547|   47149|   28161|
|Memory           |        -|      -|       -|       -|
|Multiplexer      |        -|      -|       -|      36|
|Register         |        -|      -|    3177|       -|
+-----------------+---------+-------+--------+--------+
|Total            |       13|    547|   50326|   28203|
+-----------------+---------+-------+--------+--------+
|Available        |     2940|   3600|  866400|  433200|
+-----------------+---------+-------+--------+--------+
|Utilization (%)  |    ~0   |     15|       5|       6|
+-----------------+---------+-------+--------+--------+
```

If instead I compute the multiplier limit directly in the new function with
```
 return ceil( float(n_mult) / float(CONFIG_T::reuse_factor) );
```

I get the following resource usage:

```
+-----------------+---------+-------+--------+--------+
|       Name      | BRAM_18K| DSP48E|   FF   |   LUT  |
+-----------------+---------+-------+--------+--------+
|DSP              |        -|      -|       -|       -|
|Expression       |        -|      -|       0|       6|
|FIFO             |        -|      -|       -|       -|
|Instance         |       13|    547|   46759|   28141|
|Memory           |        -|      -|       -|       -|
|Multiplexer      |        -|      -|       -|      36|
|Register         |        -|      -|    3177|       -|
+-----------------+---------+-------+--------+--------+
|Total            |       13|    547|   49936|   28183|
+-----------------+---------+-------+--------+--------+
|Available        |     2940|   3600|  866400|  433200|
+-----------------+---------+-------+--------+--------+
|Utilization (%)  |    ~0   |     15|       5|       6|
+-----------------+---------+-------+--------+--------+
``` 

The only difference is that in this case one of the three conv_1d layers in the model takes 390 fewer FF and 20 fewer LUTs.  And it takes exactly the same as if I pass the multiplier limit directly through the unused n_zeros configuration option:
```
const int multiplier_limit = CONFIG_T::n_zeros;
```

So somehow we end up with slightly different resource usage if we leave the `ceil( float(compute_n_mult<CONFIG_T>(weights)) / float(CONFIG_T::reuse_factor) )` to the main conv_1d code, but it's not that it's doing it in firmware -- there is no latency change or DSP change, which you think you would have from e.g. doing a division.

So my proposal is to do that latter -- computing the multiplier limit rather than the multiplier count in the function -- because it takes slightly fewer resources... for unknown reasons.  PR on the way

nhanvtran, 2018-06-21:
looks like we can close this since the PR is merged

closed 2018-06-21 by nhanvtran