fastmachinelearning/hls4ml #59: Memory problems when synthesizing Conv1D models

jmduarte, 2018-04-16:
@ejk43 We're trying to synthesize some larger Conv1D models in Vivado HLS 2017.2 and we're wondering if the problems we're seeing are just due to the memory available on our computer.

When we synthesize a very small Conv1D model, it works. Here's a `keras-config.yml`
```
KerasJson: example-keras-model-files/KERAS_conv1d_small.json
KerasH5:   example-keras-model-files/KERAS_conv1d_small_weights.h5
OutputDir: my-hls-test
ProjectName: myproject
XilinxPart:  xcku115-flvf1924-2-i
ClockPeriod: 5

IOType: io_parallel # options: io_serial/io_parallel
ReuseFactor: 1
DefaultPrecision: ap_fixed<16,6>
```
and the command to build it:
```
python keras-to-hls -c keras-config.yml
cd my-hls-test
vivado_hls -f build_prj.tcl
```

but when we run a larger model, e.g.
```
KerasJson: example-keras-model-files/KERAS_conv1d.json
KerasH5:   example-keras-model-files/KERAS_conv1d_weights.h5
OutputDir: my-hls-test
```
We get the following error during csynth:
```
ERROR: [XFORM 203-504] Stop unrolling loop 'ConvOut' (/home/jduarte1/hls-fpga-machine-learning/nnet_utils/nnet_conv.h:79) in function 'nnet::conv_1d<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config2>' because it may cause large runtime and excessive memory usage due to increase in code size. Please avoid unrolling the loop or form sub-functions for code in the loop body.
ERROR: [HLS 200-70] Pre-synthesis failed.
```

Is this just a memory problem? Or can we solve this by changing our HLS code? What kind of computer are you using to synthesize and compile firmware?

Thanks!

ejk43, 2018-04-17:
Seems like I've maxed out my 16 GB and eat up 8 GB of swap... I'll just let it finish and see what happens :)

ejk43, 2018-04-17:
Okay, it synthesized for perhaps 45 minutes, then hit a bunch of errors like this and eventually died:
```
@I [XFORM-501] Unrolling loop 'ConvOut' (/home/ejk/work/hls-fpga-machine-learning/nnet_utils/nnet_conv.h:79) in function 'nnet::conv_1d<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config3>' completely.
@W [XFORM-504] Stop unrolling loop 'ConvOut' (/home/ejk/work/hls-fpga-machine-learning/nnet_utils/nnet_conv.h:79) in function 'nnet::conv_1d<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config3>' because it may cause large runtime and excessive memory usage due to increase in code size. Please avoid unrolling the loop or form sub-functions for code in the loop body.
@I [XFORM-501] Unrolling loop 'ConvFilt' (/home/ejk/work/hls-fpga-machine-learning/nnet_utils/nnet_conv.h:80) in function 'nnet::conv_1d<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config3>' completely.
@W [XFORM-504] Stop unrolling loop 'ConvFilt' (/home/ejk/work/hls-fpga-machine-learning/nnet_utils/nnet_conv.h:80) in function 'nnet::conv_1d<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config3>' because it may cause large runtime and excessive memory usage due to increase in code size. Please avoid unrolling the loop or form sub-functions for code in the loop body.
@I [XFORM-501] Unrolling loop 'ConvChan' (/home/ejk/work/hls-fpga-machine-learning/nnet_utils/nnet_conv.h:81) in function 'nnet::conv_1d<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config3>' completely.
@I [XFORM-501] Unrolling loop 'ConvMult' (/home/ejk/work/hls-fpga-machine-learning/nnet_utils/nnet_conv.h:82) in function 'nnet::conv_1d<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config3>' completely.
/home/ejk/Xilinx/Vivado_HLS/2015.4/bin/loader: line 164: 18843 Killed                  "$RDI_PROG" "$@"
```

So,  looks like I see a similar issue too. I *guess* it's perhaps memory-limitations... I'll look at it some more. I havent seen something like that before, but I have also not tried to synthesize a network with quite so many parameters

ejk43, 2018-04-18:
Yikes... that's quite the error message... `Unsupported enormous number of load/store instructions`

Here's what I could find on the forum: https://forums.xilinx.com/t5/SDAccel/Load-store-instructions-crash-ERROR-XFORM-203-1403/m-p/752313/highlight/true#M605

Sad to see the current answer is less unrolling. Any idea what the magnitude of multipliers works out to be for your networks (and whether they should fit into the FPGA unrolled)?

nhanvtran, 2018-04-18:
Hm, do you think it's a real physical limitation to the FPGA?  Like ... ok you tried to unroll all these multiplications at the same time (like 10000) but there's no way to physically do this on the FPGA the way the DSPs are laid out?  

Or is it an issue with the HLS compiler at some higher level abstraction?

benjaminkreis, 2018-04-18:
The numbers I reported above and below here are for a Virtex 7 690T with 3.6k DSPs.

We get the same `large runtime and excessive memory` with reuse factors that bring the total number of DSPs below the number on the board.  For example 16->200->200->32->5 with reuse=50 and <18,8> should be ~1k DSPs.

In the case of `Unsupported enormous number of load/store instructions` with reuse=20 reported above, it was 16->100->100->32->5, so about 750 DPSs.

This doesn't rule out lay-out, but to me it seems more like abstract HLS issues.  I don't know what load/store instructions are, but the error about `large runtime and excessive memory` seems more clear.

nhanvtran, 2018-04-18:
Ok makes sense...sorry I didn't get the chance to install 2018.1 today, I'll do tomorrow.  maybe if we get some direct discussion with the Xilinx people we can ask them directly

jmduarte, 2018-05-25:
Starting PR #62 to address this...

benjaminkreis, 2019-03-12:
An attempt at summarizing status of Conv1D and Conv2D examples using Vivado 2018.2::

Conv1D
====================================================
KERAS_conv1d.json
- larve conv1d, but actually hits bug related to sublayer of dense layer

KERAS_conv1d_small.json  
- synthesizes ok

KERAS_conv1d_small_nfilt5.json  
- synthesizes ok

*New* conv1d example
- https://www.dropbox.com/sh/onghdkm8oh40bwg/AACPJowZxHD9Ozb_1Rkx6Aqva?dl=0
- loop unroll error

Conv2D
====================================================
KERAS_conv2d_model.json 
- loop unroll error
- synthesizes ok in Vivado 2017.2

jetTagger_Conv2D_Small.json
- loop unroll error

jetTagger_Conv2D_Small_NoBatchNorm.json
-  loop unroll error

closed 2022-06-19 by jmduarte