fastmachinelearning/hls4ml #70: Concatenate layer

jmduarte, 2018-07-05:
Add support for `Concatenate` layer type:

https://github.com/keras-team/keras/blob/master/keras/layers/merge.py#L320

vloncar, 2018-07-20:
I have been looking at the the possible ways to implement this, however, other than the simple approach demonstrated in `nnet::merge` function which is used in sublayer project the other approaches don't work nicely with HLS. Ideally, we would have a `nnet::concatenate` function which takes a list of arguments, but when creating a list of existing arrays, e.g., for 1D arrays:

```
layer_t logits_0[N_LAYER/2];
layer_t logits_1[N_LAYER/2];
layer_t (*logits_list[2])[N_LAYER/2] = {&logits_0, &logits_1};
```
HLS  synthesis complains that using `&` results in an unsupported access pattern and thus the arrays cannot be partitioned (which is bad). Using pure pointer approach has the same problems.

Another approach I tried is to use function with variadic arguments (similar to C `printf`), but this crashes the synthesis process (it segfaults on a line where I access a pointer through `va_arg`).

This leaves only the simple approach with multiple functions with different number of arguments. So we would have one `nnet::concatenate` which concatenates 2 arrays, one for 3 arguments etc. The same for other Keras merge layers and (eventually) the same 2D, 3D...

Handling this in keras-to-hls is not a problem, so I am thinking about doing it this way even though it is not perfect. Thoughts?

benjaminkreis, 2018-07-20:
Hi @vloncar, thanks for investigating this!  I think going with the `nnet::merge` is good.   Maybe not the most beautiful, but simple is good for HLS.

closed 2021-09-12 by jmduarte