fastmachinelearning/hls4ml #79: Syn failed for 3Layer with sublayer

zhenbinwu, 2018-07-17:
The current master branch of the hls4ml can't synthesize  the 3layer Model. 
```
ERROR: [XFORM 203-103] Array 'mult.V' (/data/benwu/HLS4ML_2018/hls4ml/nnet_utils/nnet_layer.h:56): partitioned elements number (2048) has exeeded the threshold (1024), which may cause long run-time.
ERROR: [HLS 200-70] Pre-synthesis failed.
command 'ap_source' returned error code
    while executing
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "

INFO: [Common 17-206] Exiting vivado_hls at Fri Jul 13 12:00:25 2018...
```
The error can be trace to https://github.com/hls-fpga-machine-learning/hls4ml/pull/62/files#diff-b37c065f136460b015788b96b5c25102L52

jmduarte, 2018-07-18:
Somehow the issue comes from re-writing the `mult` array as a 1D array instead of a 2D array: 

https://github.com/hls-fpga-machine-learning/hls4ml/commit/d90e74a7c92eb63e675e27396177cb05b9286f00#diff-b37c065f136460b015788b96b5c25102L96 
https://github.com/hls-fpga-machine-learning/hls4ml/commit/d90e74a7c92eb63e675e27396177cb05b9286f00#diff-b37c065f136460b015788b96b5c25102L96 

Conversely if I comment out the `mult` array partitioning (much like the `weight` array is *not* partitioned), it also works again (although I see some other warnings)
https://github.com/hls-fpga-machine-learning/hls4ml/blob/d90e74a7c92eb63e675e27396177cb05b9286f00/nnet_utils/nnet_layer.h#L70

Maybe it would help if I knew why the `weight` array partitioning was commented out. It looks like @benjaminkreis did it?

nhanvtran, 2018-07-18:
just to record this for everyone...

So this error actually looks like a difference in Vivado version.  This does not occur with 2017.2.  It looks like Xilinx added an array size limit for fully partitioning in 2018.2 -- probably because of our complaints about RAM  limits in the compiler 😱

closed 2021-09-12 by jmduarte

SteCla0, 2022-06-23:
Error still exists with 16x100x100x100x100x5 example in hls4ml v0.6.0 and vivado 2020.1