fastmachinelearning/hls4ml #80: Modification for Quartus

nhanvtran, 2018-07-19:
adapt hls4ml (Vivado based) to be also be compatible with Quartus

nhanvtran, 2019-06-06:
https://github.com/vloncar/hls4ml/tree/intel

kutenai, 2020-04-04:
Has some work been done toward this goal? I'm wondering what is involved to use this on an intel/altera FPGA.

nhanvtran, 2020-04-06:
hi @kutenai -- there is work on-going to integrate intel/altera FPGAs into the workflow.  It was a bit slow-going at first, but there is more progress recently.  we don't have a fixed timeline for any release, but we could try to put out something in pre-alpha on the timescale of ~weeks/month or so

kutenai, 2020-04-06:
That is very exciting news... thanks for the work. 
I'm not sure what is involved, or if there is a way I could help out or not.. but, I could theoretically do that in a month or so. Right now I'm in a crunch to implement a CNN in a Cyclone V, so using OpenCL, but this hls4ml looks very *very* impressive... and I daresay like a better long-term solution.

kutenai, 2020-05-29:
Are there any updates on this @nhanvtran ?

hamzajaved780, 2020-06-02:
How far did you get with your CNN implementation? As of now, we only support Quantized Dense networks with user-defined latency/area trade off. We do plan to extend this to CNN's in the near future. Also what specific models are you interested in implementing?

kutenai, 2020-06-12:
@hamzajaved780 -- I did not get far at all. I've been trying to use OpenCL with a Cyclone V device, and HLS -- lots of buzzwords in there ;-)

But, I just found out from Intel that this is not supported. It used to be, but it never really panned out I guess, so support was dropped in Quartus 17. I've been trying to make this work in Quartus 18, and have had no luck at all.. now, I know why.

I'm coming back to look at HLS...

I have one *SUPER* important question that I hope someone can answer... Does hls4ml outputs some code that I could *manually* import into a Quartus project?

I'm find doing the legwork on that, and would even be interested in porting that back in for support if possible, but I would like to know if I'm on a "reasonable" track.

My hope is that the "Xilinx" support is more along the lines of intergartion into the IDE, generation of .tcl files, etc... but the code itself is *close* to portable. If so, I'll figure out the steps to get it into Quartus... 

Sound possible?

hamzajaved780, 2020-06-13:
I think you can import both the HLS and the compiled project into Quartus, the HLS compiler even generates a Quartus project file. 

CNN implementation on the Xilinx side is still a work in progress, mainly limited by the size of layer/model you want to implement. 

Portability wont be an issue, the HLS code is mostly shared between the two, although you'll have to do a lot of optimizations to get similar performance (mainly through pragmas and interface modification).

I'll be starting my work on implementing CNNs from next week, so it'd be nice if you have some specific model that we can target for this device and see how well it performs.

My recommendation would be to start from the new large dense implementation and then convert it into a Conv layer (using the Xilinx Conv template). If we can get something working, generalizing it would be trivial. Limiting resource usage is super critical when dealing with millions of MACs (Multiplication and Addition), so we might have to change a bit of architecture for it to work well on Intel.

kutenai, 2020-06-13:
I don't see in the "example-models" projects where it generates a quartus output file.
I ran the example `hls4ml conviert -c keras-config.yml`.

It ran fine, but there is no quartus output file. Note that this is on the `master` branch.

hamzajaved780, 2020-06-13:
You'll have to run "make myproject-fgpa" for it to generate those files. Also, master only works with Vivado. Try this repo instead: https://github.com/hamzajaved780/hls4ml/tree/activation-update. I'll update this repo but it should give you a nice idea.

kutenai, 2020-06-13:
I noticed  you did not fork your repo, that's too bad, as it does not allow comparing against the main repo (as easily), or merging to the man repo with a PR.. however, it's great the work your doing!!!!

hamzajaved780, 2020-06-14:
The main repo uses an entirely different frontend than the Quartus one, it'll be updated however soon.

pierinim, 2020-06-14:
hls4ml will eventually consisted of three frontends for three  different HLS backends. When these developments will be over, all these frontends will be integrated in a single package.

jmduarte, 2022-06-19:
Close with #245

closed 2022-06-19 by jmduarte