fastmachinelearning/hls4ml #82: Zero initializing the arrays

vloncar, 2018-07-25:
While fiddling with @jngadiub's [binary network](https://github.com/jngadiub/BinaryDense_HLS_proj) I have stumbled upon trivial change which seems to reduce resource usage and latency significantly.

In `compute_layer_nobias`, array [`acc`](https://github.com/jngadiub/BinaryDense_HLS_proj/blob/master/nnet_utils/nnet_layer.h#L139) is initialized to zero in a [loop](https://github.com/jngadiub/BinaryDense_HLS_proj/blob/master/nnet_utils/nnet_layer.h#L182). Using array initialization syntax `{0}`:
```
typename CONFIG_T::accum_t acc[CONFIG_T::n_out] = {0};
```
and commenting out/removing `ResetAccum` loop reduces the resource usage and latency significantly. This is still valid C++ code, it passes csim with the same results, synthetises, and even cosim passes. The question is why does it use significantly less resources?

Here are the examples of synthesis reports without array initialization and with array initialization.
Loop initialization:
[image]

Array initialization:
[image]

Looks too good to be true :-)

The same idea can be applied elsewhere, e.g,, in `softmax` activation. It doesn't change the latency there, but reduces resource usage.

benjaminkreis, 2018-07-25:
Hm yes there is something wrong with this, because the DSP usage really cannot drop to 4, for example.

I used to use `={0}` to initialize arrays in the test bench, but I removed this after noticing some problems.  I can't remember exactly what the problem was -- I think maybe my output was changing run to run, suggesting that the array wasn't actually initialized to all zeros.  

I'm surprised it passes csim and cosim with the same results.  How many samples did you run the inference on?

jngadiub, 2018-07-25:
I am confused how can it be that the batch normalization layers use 1 DSP to do up to 64 multiplications of the kind ap_int<8>*ap_fixed<18,8>, with reuse factor = 1.

vloncar, 2018-07-25:
@benjaminkreis  Perhaps you are refering to the issue mentioned [here](https://forums.xilinx.com/t5/Vivado-High-Level-Synthesis-HLS/HLS-bug-Array-initialisation/td-p/683776)? Supposedly it has since been solved (I also found that out in the forums, but cant find the thread now). But in our case the outputs match. I have just used Jen's default inputs.

I agree that this is probably wrong, but I would like to understand why, and avoid this and any similar HLS ~~bugs~~ features in the future.

benjaminkreis, 2018-07-25:
Just to add more about the failure, if you look at the ports of the compute_layer vhdl blocks, they don't even take in the inputs or output the result:

```
entity compute_layer_nobias_1 is
port (
    data_0_V_read : IN STD_LOGIC_VECTOR (1 downto 0);
    ap_return : OUT STD_LOGIC_VECTOR (4 downto 0) );
end;
```

```
entity compute_layer_nobias_2 is
port (
    ap_clk : IN STD_LOGIC;
    ap_rst : IN STD_LOGIC;
    data_V_read : IN STD_LOGIC_VECTOR (287 downto 0);
    ap_return : OUT STD_LOGIC_VECTOR (17 downto 0);
    ap_ce : IN STD_LOGIC );
end;
```

```
entity compute_layer_nobias_3 is
port (
    data_0_V_read : IN STD_LOGIC_VECTOR (1 downto 0);
    ap_return : OUT STD_LOGIC_VECTOR (2 downto 0) );
end;
```

and it's similar for the normalize layers.  So it's certain it's failing.

I still don't know why exactly this happens, but we should avoid `={0}`.

vloncar, 2018-08-06:
> @vloncar <https://github.com/vloncar> I think we can close this. Do you

Yes, please close it. This was just an anomaly I wanted to understand
better, not an actual issue with any current code.

closed 2018-08-06 by benjaminkreis