fastmachinelearning/hls4ml #84: Synthesis fails with io_serial

vloncar, 2018-07-26:
After changing `IOType` to `io_serial` in `keras-config.yml` the resulting code does not synthesize. Only 1layer with sigmoid activation succeeds, and that requires #81 to succeed.

Another problem is that models with Conv layers produce code with the wrong pragmas, e.g., `#pragma HLS STREAM variable=logits1 complete depth=1`. This fails with the error:
```
ERROR: [HLS 200-70] pragma 'STREAM variable=logits1 complete depth=1' has unknown option 'complete'
ERROR: [HLS 200-70] '#pragma HLS STREAM variable=logits1 complete depth=1' is not a valid pragma.
```
However, this is trivially solved by removing the offending keyword. I can make a PR with the change in `hls-writer.py`, but this does not solve all the problems of synthesis hanging.

The errors are different for different models, some take forever to finish (like 3layer), some just fail with some error about memory (sorry, I didn't keep the logs), and some (like KERAS_conv1d_small model) fails with a an error like:
```
WARNING: [XFORM 203-124] Array  'conv_layer2_in.V' (firmware/serial_test.cpp:74): The entries are not accessed in sequential order.
WARNING: [XFORM 203-124] Array  'conv_layer3_in.V' (firmware/serial_test.cpp:87): The entries are not accessed in sequential order.
WARNING: [XFORM 203-124] Array  'logits5.V' (firmware/serial_test.cpp:105): The entries are not accessed in sequential order.
ERROR: [XFORM 203-123] Cannot stream  'data.V' (firmware/serial_test.cpp:42): The entries are not accessed in sequential order.
ERROR: [HLS 200-70] Pre-synthesis failed.
```

nhanvtran, 2018-08-02:
hi @vloncar --  a little bit of history is that serial mode was a capability that we did not support as much.  we will have to reconcile this with the conv models.  i'm moving this to our to-do list

ejk43, 2018-08-03:
Hi guys, I'm planning to (finally) take a crack at serial mode this month, if that doesn't conflict with anyone else's plans. I'm still quite interested in running serially

nhanvtran, 2018-08-03:
hi @ejk43 -- there are no conflicts here, you are certainly welcome to take a crack at this!

ejk43, 2018-08-22:
Starting to unpack what's going on here, so I thought I'd capture a few thoughts. My plan is to start with dense layers and move towards conv layers after dense is OK across the suite of example projects. Let me know if there's specific areas that would be good to focus on, otherwise I'll just keep working through dense layers first.

It appears serial mode is "close" to working on dense layers with the 1layer keras project. For comparison, in parallel mode, I see nominal resource usage as expected...
```
+-----------------+---------+-------+--------+--------+
|       Name      | BRAM_18K| DSP48E|   FF   |   LUT  |
+-----------------+---------+-------+--------+--------+
|DSP              |        -|      -|       -|       -|
|Expression       |        -|      -|       0|       6|
|FIFO             |        -|      -|       -|       -|
|Instance         |        1|    305|    8433|   11777|
|Memory           |        -|      -|       -|       -|
|Multiplexer      |        -|      -|       -|      36|
|Register         |        -|      -|    1365|       -|
+-----------------+---------+-------+--------+--------+
|Total            |        1|    305|    9798|   11819|
+-----------------+---------+-------+--------+--------+
|Available        |     2940|   3600|  866400|  433200|
+-----------------+---------+-------+--------+--------+
|Utilization (%)  |    ~0   |      8|       1|       2|
+-----------------+---------+-------+--------+--------+
```

For serial mode,  DSP usage is perfect -- I expect 32 DSPs for the hidden layer, and 1 DSP for the output layer. I also expect the sigmoid output occupies 1 BRAM, which it does. FFs look okay, but LUTs are embarrassingly high:

```
+-----------------+---------+-------+--------+--------+
|       Name      | BRAM_18K| DSP48E|   FF   |   LUT  |
+-----------------+---------+-------+--------+--------+
|DSP              |        -|      -|       -|       -|
|Expression       |        -|      -|       0|       4|
|FIFO             |        0|      -|      15|      89|
|Instance         |        1|     33|   11519|  297386|
|Memory           |        -|      -|       -|       -|
|Multiplexer      |        -|      -|       -|       -|
|Register         |        -|      -|       -|       -|
+-----------------+---------+-------+--------+--------+
|Total            |        1|     33|   11534|  297479|
+-----------------+---------+-------+--------+--------+
|Available        |     2940|   3600|  866400|  433200|
+-----------------+---------+-------+--------+--------+
|Utilization (%)  |    ~0   |   ~0  |       1|      68|
+-----------------+---------+-------+--------+--------+
```

I'm not sure what would instigate such an outrageous amount of LUTs. It appears there's no BRAMs used for the intermediate layers; But still, LUT count is manageable in parallel mode, which does not use BRAMs either. I'm suspicious that flattening the weights may have disrupted the BRAM storage, since there's still a partition directive applied to the array... Perhaps a solution here could be array_reshape (or array_partition) with a cyclic factor=NOUT? Anyone tried this successfully before?

ejk43, 2018-08-24:
It seems like cyclic partition mostly does the trick for dense-layer serial mode (https://github.com/hls-fpga-machine-learning/hls4ml/tree/ejk/serial-fix-dense). Pretty much everything else appeared intact and functional out of the box! Based on a few initial experiments with the 1-layer and 3-layer examples, I can get resources and II to behave pretty well across a selection of reuse_factors.

For example: 1-layer model with resource reuse 4x on the hidden layer. Total DSP usage = 8 (32/4 outputs in hidden layer) + 1 (one output in second layer)
```
+-----------------+---------+-------+--------+--------+
|       Name      | BRAM_18K| DSP48E|   FF   |   LUT  |
+-----------------+---------+-------+--------+--------+
|DSP              |        -|      -|       -|       -|
|Expression       |        -|      -|       0|       4|
|FIFO             |        0|      -|      15|      89|
|Instance         |        1|      9|    2684|    5805|
|Memory           |        -|      -|       -|       -|
|Multiplexer      |        -|      -|       -|       -|
|Register         |        -|      -|       -|       -|
+-----------------+---------+-------+--------+--------+
|Total            |        1|      9|    2699|    5898|
+-----------------+---------+-------+--------+--------+
|Available        |     2940|   3600|  866400|  433200|
+-----------------+---------+-------+--------+--------+
|Utilization (%)  |    ~0   |   ~0  |   ~0   |       1|
+-----------------+---------+-------+--------+--------+
```

3-layer model with default reuse 1x:
```
   +-----+-----+-----+-----+----------+
    |  Latency  |  Interval | Pipeline |
    | min | max | min | max |   Type   |
    +-----+-----+-----+-----+----------+
    |  106|  106|   86|   86| dataflow |
    +-----+-----+-----+-----+----------+
+-----------------+---------+-------+--------+--------+
|       Name      | BRAM_18K| DSP48E|   FF   |   LUT  |
+-----------------+---------+-------+--------+--------+
|DSP              |        -|      -|       -|       -|
|Expression       |        -|      -|       0|       4|
|FIFO             |        0|      -|      35|     207|
|Instance         |        3|    133|   10339|   21858|
|Memory           |        -|      -|       -|       -|
|Multiplexer      |        -|      -|       -|       -|
|Register         |        -|      -|       -|       -|
+-----------------+---------+-------+--------+--------+
|Total            |        3|    133|   10374|   22069|
+-----------------+---------+-------+--------+--------+
|Available        |     2940|   3600|  866400|  433200|
+-----------------+---------+-------+--------+--------+
|Utilization (%)  |    ~0   |      3|       1|       5|
+-----------------+---------+-------+--------+--------+
```

I think it's also kind of interesting to demonstrate a "well-tuned" serial model... If I set the reuse_factor to 2 on the first and third hidden layers, I can reduce DSP usage by about 50 DSPs for this model without sacrificing logic, initiation interval, or throughput. Neat!
```
    +-----+-----+-----+-----+----------+
    |  Latency  |  Interval | Pipeline |
    | min | max | min | max |   Type   |
    +-----+-----+-----+-----+----------+
    |  110|  110|   86|   86| dataflow |
    +-----+-----+-----+-----+----------+
+-----------------+---------+-------+--------+--------+
|       Name      | BRAM_18K| DSP48E|   FF   |   LUT  |
+-----------------+---------+-------+--------+--------+
|DSP              |        -|      -|       -|       -|
|Expression       |        -|      -|       0|       4|
|FIFO             |        0|      -|      35|     207|
|Instance         |        3|     85|   10279|   22978|
|Memory           |        -|      -|       -|       -|
|Multiplexer      |        -|      -|       -|       -|
|Register         |        -|      -|       -|       -|
+-----------------+---------+-------+--------+--------+
|Total            |        3|     85|   10314|   23189|
+-----------------+---------+-------+--------+--------+
|Available        |     2940|   3600|  866400|  433200|
+-----------------+---------+-------+--------+--------+
|Utilization (%)  |    ~0   |      2|       1|       5|
+-----------------+---------+-------+--------+--------+
```

One remaining question for the dense layer is that the weights appear to be instantiated as logic instead of BRAM -- the only BRAMs used now are from the sigmoid output. I feel like we should be able to get these weights into BRAM to cut down some of the extra LUTs...

nhanvtran, 2018-08-29:
hi @ejk43 thanks for going through this -- the results are really nice!  i ran through the dense 3 layer model and get similar results to what you have seen.  This per-layer reuse factor is really cool :)...

A few comments:
- the latency looks like it's driven by the softmax which might be undesirable.  do I understand this is just 5 times the latency of softmax? 
- in this scheme of serial mode there's not really a way to take advantage of pruned models (there's a small amount of resource gain from using pruned mode).  do you have any ideas on how to generically incorporate this?
- include thing BRAMs is really a nice functionality that I think we'd like to have.  would you be interested in looking at this next?  i still think it's worth merging this first to get something in that works.

nhanvtran, 2018-08-29:
One more thing is once this is merged, we might want to put up a nice explanation/illustration for how serial mode is implemented in the documentation (basically 1 DSP per layer output node) because there could be many different ways to do this.

ejk43, 2018-08-30:
> the latency looks like it's driven by the softmax which might be undesirable. do I understand this is just 5 times the latency of softmax?

Softmax is interesting because it seems to break the conventional HLS streaming paradigm with its double for-loop. When handling streaming data, each iteration of the loop can only access a single element, and it must be accessed in array order. Since softmax breaks these rules, I suspect HLS needs to buffer up the entire input array before processing. This is relatively straightforward for the 3-layer model since there's just 5 elements. I would assume the total latency is (softmax latency) * (input size) but not totally sure where HLS is getting the latency output.

I'll also add: currently RTL simulation of softmax is broken in serial mode. I cannot complete a RTL simulation when softmax activation is in the model -- it hangs perpetually with both Vivado 2017.4 and 2018.2. Though if I replace with a different layer that is more serial-compatible, then RTL simulation works OK. 

> in this scheme of serial mode there's not really a way to take advantage of pruned models (there's a small amount of resource gain from using pruned mode). do you have any ideas on how to generically incorporate this?

I seem to recall thinking about this before... It occurs to me now that the idea we're searching for is FPGA-based sparse matrix multiplication, which seems like should be a thing that people have studied. This paper looks insightful, or at least a good start: https://pdfs.semanticscholar.org/8a4d/9bc5131a074e67594b9361fbf75d8e35c55e.pdf

Sounds like a sparse matrix multiply probably requires a dedicated dense-layer function since the architecture strays pretty far from a basic matrix multiply. 

> include thing BRAMs is really a nice functionality that I think we'd like to have. would you be interested in looking at this next? i still think it's worth merging this first to get something in that works.

Sure, I'll try to get that working. I really seem to recall having BRAMs working in my old rfnoc version,  however I've had some similar issues getting BRAMs to instantiate correctly in a different project recently.

nhanvtran, 2018-08-30:
> This is relatively straightforward for the 3-layer model since there's just 5 elements. I would assume the total latency is (softmax latency) * (input size) but not totally sure where HLS is getting the latency output.

Yeah this was my guess as well - this makes sense to me.

> I'll also add: currently RTL simulation of softmax is broken in serial mode. I cannot complete a RTL simulation when softmax activation is in the model -- it hangs perpetually with both Vivado [...]

I did see this too.  I feel like this is something to resolve before merging.  Do you have any thoughts on how to debug this?  

Could we think about only allowing softmax in parallel mode?  We've also thought about more firmware friendly adaptions of softmax in the past. 

> This paper looks insightful, or at least a good start: https://pdfs.semanticscholar.org/8a4d/9bc5131a074e67594b9361fbf75d8e35c55e.pdf
> Sounds like a sparse matrix multiply probably requires a dedicated dense-layer function since the architecture strays pretty far from a basic matrix multiply.

Hm interesting.  I'll try to have a deeper look.  In the meantime, a facsimile that we've had in mind is working towards parallel mode with high reuse factor and weights in BRAMs

> Sure, I'll try to get that working. I really seem to recall having BRAMs working in my old rfnoc version, however I've had some similar issues getting BRAMs to instantiate correctly in a different project recently.

Yeah I remember it was in the original version too!

ejk43, 2018-08-30:
> currently RTL simulation of softmax is broken in serial mode

I think I've resolved the softmax problem in serial mode: https://github.com/hls-fpga-machine-learning/hls4ml/commit/f77c28596d7e473083b37f03ce156817266c6f04

Turns out that HLS really did not like accessing array elements in non-sequential order. The fix was to create an intermediate cache which accesses the input array in sequential order.

I believe this change should be compatible with parallel mode, though I havent had positive confirmation yet -- Any suggestions which model to test? I didnt want to use 1layer, since it defaults to a single output. Im trying to synthesize the KERAS_3layer model, and I keep hitting the error that `partitioned elements number (2048) has exeeded the threshold (1024), which may cause long run-time` even for reuse factors > 1 (both Vivado 2017.4 and 2018.2).

Overall, my (admittedly noncomprehensive) tests so far indicate serial mode is working for both dense layers and activation functions. May be worth opening a PR for this branch in the meantime and move on to BRAMs or other serial mode operations.

nhanvtran, 2018-08-30:
Great!  Try 3 layer with 2017.2 -- it was working there if you have it around.

In the meantime, we're trying to find the right syntax in the tcl script to configure the array partition max size to be > 1024 in 2018.2

Once it's tested, I agree to merge this and deal with BRAM in a different PR

closed 2018-09-05 by nhanvtran