fastmachinelearning/hls4ml #90: testing MNIST model from Grzegorz Korcyl

nhanvtran, 2018-08-28:
see if we can synthesize this with the master

nhanvtran, 2018-08-30:
It's weird, I thought people were getting the mails from Grzegorz.  In any case, I copied them to dropbox:
https://www.dropbox.com/s/8689nfpykw5j945/mod_15_15.h5?dl=0
https://www.dropbox.com/s/ju7fy2pvz8nhcdl/mod_15_15.json?dl=0

ejk43, 2018-08-31:
As a data point, this model seems to work pretty well in serial mode and passes synthesis/RTL sim. The II reflects the input layer size (784 features need to get clocked in, one feature at a time). See branch here: https://github.com/hls-fpga-machine-learning/hls4ml/tree/ejk/test-grzegorz-model

```
+-----+-----+-----+-----+----------+
|  Latency  |  Interval | Pipeline |
| min | max | min | max |   Type   |
+-----+-----+-----+-----+----------+
|  804|  804|  789|  789| dataflow |
+-----+-----+-----+-----+----------+
+-----------------+---------+-------+--------+--------+
|       Name      | BRAM_18K| DSP48E|   FF   |   LUT  |
+-----------------+---------+-------+--------+--------+
|DSP              |        -|      -|       -|       -|
|Expression       |        -|      -|       0|       4|
|FIFO             |        0|      -|      25|     148|
|Instance         |       18|     40|    3759|    7771|
|Memory           |        -|      -|       -|       -|
|Multiplexer      |        -|      -|       -|       -|
|Register         |        -|      -|       -|       -|
+-----------------+---------+-------+--------+--------+
|Total            |       18|     40|    3784|    7923|
+-----------------+---------+-------+--------+--------+
|Available        |     2940|   3600|  866400|  433200|
+-----------------+---------+-------+--------+--------+
|Utilization (%)  |    ~0   |      1|   ~0   |       1|
+-----------------+---------+-------+--------+--------+
```

One thing I found was that the hls-writer partitions model into sublayers by default. Based on my read of the sublayer code, it looks like the sublayer content is more or less trying to fool the HLS compiler into working on bigger models? Unfortunately the sublayer paradigm doesnt look like it will apply cleanly onto serial mode -- I'd suggest we simply disable the sublayer partitioning when using io_serial.

closed 2018-11-29 by nhanvtran