fastmachinelearning/hls4ml #105: Conv2d image shapes

benjaminkreis, 2018-10-04:
I don't fully understand the problem yet, but we have a bug in defining the image shapes of Conv2ds.

I have a model where the height and width of a second Conv2d is being assigned the height and width of the first Conv2d.

Relevant code:
https://github.com/hls-fpga-machine-learning/hls4ml/blob/master/keras-to-hls/keras-to-hls.py#L234-L235
https://github.com/hls-fpga-machine-learning/hls4ml/blob/master/keras-to-hls/keras-to-hls.py#L280 

I'm investigating

jmduarte, 2018-10-05:
I think this line:

https://github.com/hls-fpga-machine-learning/hls4ml/blob/master/keras-to-hls/keras-to-hls.py#L262

should be something like this:
```
current_shape = [current_shape[0], layer['out_width'], layer['out_height'], layer['n_filt']]
```

basically it looks like we're not updating the `current_shape` after a Conv2D layer

benjaminkreis, 2018-10-05:
Indeed, that is the line!  Thank you, @jmduarte!

Will make a PR, just pulling this line back one tab to run for both valid and same padding:
https://github.com/hls-fpga-machine-learning/hls4ml/blob/master/keras-to-hls/keras-to-hls.py#L272

well spotted