fastmachinelearning/hls4ml #118: Compiler errror with large amount of weights

zhenbinwu, 2018-11-10:
This issue was originally reported by Rishraj. He has a NN with layer 1 dimensions 784x512, layer 2 512x512, layer 3 512*512 and layer4 512*10. But the code failed at CSIM compiling:
> INFO: [SIM 211-2] *************** CSIM start ***************
 INFO: [SIM 211-4] CSIM will launch GCC as the compiler.
   Compiling ../../../../myproject_test.cpp in debug mode
   Compiling ../../../../firmware/myproject.cpp in debug mode
gcc: internal compiler error: Segmentation fault (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
make: *** [obj/myproject.o] Error 4
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
INFO: [SIM 211-3] *************** CSIM finish ***************

This is track down to the weight initiation. For our Dense 200 Network, we have only 164,200 weights and passed the CSIM, while the above network has 930,796 weights.

From a series of test, the vivado can compile 262,144 weights , but failed with 663,168 weights. The memory usage is ~10% on correlator2.fnal.gov. This seems another HLS compiler issue.

nhanvtran, 2018-11-10:
Thanks @zhenbinwu -- is this a usual stack overflow issue? It seems like a standard gcc thing

https://stackoverflow.com/questions/1847789/segmentation-fault-on-large-array-sizes

zhenbinwu, 2018-11-11:
I think the problem is initialing the value as ap_fixed. With the default hls4ml, the weight_default_t (ap_fixed) type is used to initialize the weight. If I changed it to float type, then the code can pass
the compilation.

vloncar, 2018-11-11:
This looks like the same issue we already observed before, e.g., by Jen in [PR72](https://github.com/hls-fpga-machine-learning/hls4ml/pull/72#issuecomment-407526574). Did you try using the clang compiler or newer Vivado HLS (which uses newer GCC)?

nhanvtran, 2018-11-13:
just to follow up on this, rishiraj reported that this is fixed now by moving to the latest version of Vivado  2018.2 so i think you are right @vloncar that this is fixed now with the latest version.

there's still an issue with rishiraj's model for csynth, but we are trying to reproduce this error and will follow up more

zhenbinwu, 2018-11-20:
Following up on this issue. I am using 2018.2 with gcc compiler. @vloncar How do I switch to clang compiler in vivado_hls? I can't find the command.

The model from rishiraj stored weights as ap_int. Switching from ap_fixed to ap_int does pass the csim compile. But with ap_fixed type, I am still getting compiler crash. If you are interested, you can test it with a large weight file from [here](https://gist.github.com/zhenbinwu/8dfa79be1d4dc74a67430c5ebfe2c958)

vloncar, 2018-11-20:
@zhenbinwu If you are using `build_prj.tcl` script for vivado_hls, you can add `-compiler clang` to `csim_design` line. If you are using GUI, then you can make the same change in the project properties

zhenbinwu, 2018-11-20:
@vloncar Thanks. Let me try it

zhenbinwu, 2018-11-20:
Cool! With Clang, it is much faster and compiled without error. Maybe we should use clang as default. 
Thanks @vloncar ! I will close this issue.

closed 2018-11-20 by zhenbinwu

vloncar, 2018-11-20:
Clang in Vivado HLS is not available for Windows (though I guess it will be supported in a future release). If we switch to it as a default we lose that compatibility. Not sure if anyone cares...

zhenbinwu, 2018-11-20:
I see. We can keep it as a known solution in case this happen.