fastmachinelearning/hls4ml #120: Restructuring hls writer

benjaminkreis, 2018-11-20:
Here we can discuss restructuring [hls_writer.py](https://github.com/hls-fpga-machine-learning/hls4ml/blob/master/hls-writer/hls_writer.py).  Now that we are supporting so many layer types this code has gotten rather inelegant.  

I chatted with @jmduarte and @thesps recently, and here are a few initial thoughts:

1) hls_writer expects a list of python dictionaries that hold the layer info extracted in keras-to-hls or pytorch-to-hls.  For @sergojin's model with the output of one layer feeding into two following layers (#104), we should add to this dictionary where the inputs are coming from rather than assume the ordering in the json.  @jmduarte made the nice suggestion of making the list of dictionaries a dictionary of dictionaries so that we can access layers by a key such as an index or name rather than an index in a list.  This will make it easier to find the input to a layer.

2) Currently hls_writer's outer loop is through the lines of template C++ files.  It has an inner loop over layers at each place it needs (e.g. when including the weight files, when adding the layers in the top level, etc.).  Perhaps it would be better to make the outer loop over the layers (ie the dictionary of layer dictionaries) and append python objects holding the C++ code.  @jmduarte points out that this would make adding new layer types easier because development would be more confined to one section of code.    

3) A lot of the confusing logic in hls_writer is in the sharing of the `#define` parameters in parameters.h between layers.  This makes the generated C++ code easier for a human to read, but is making hls_writer harder to write because we have to write out every possible combination.  For example, when making a dense layer, we check which layer came before to decide if we want to write n_in as a function of a previous Conv2D output (`n_in = OUT_HEIGHT*OUT_WIDTH*N_FILT`), a previous Convn1D output (`n_in = Y_OUTPUTS*N_FILT`), or a previous dense output (`n_in = N_LAYER_1`).   Maybe we can still use `#define` parameters set by info only from the present layer dictionary (assuming this can be extracted from the json and h5),  sometimes leading to redundant ones created in the next.  This would keep some of the readability but avoid having to code the logic to explicitly handle every possible combination in hls_writer.

zhenbinwu, 2018-11-20:
For 1, this is partially started for MLP， support Pytorch,  at https://github.com/hls-fpga-machine-learning/hls4ml/tree/zw/addPytorch

vloncar, 2018-11-20:
Are you guys already working on this and/or have a specific implementation on mind, or it needs/welcomes volunteers? I had many issues with hls_writer in my unfinished attempt at merge layers (#70), and would love to tackle this. I think we can borrow some ideas from ONNX and how they do IR.

nhanvtran, 2018-11-28:
sorry for chiming in late @vloncar -- this would greatly appreciated and very very important

we can discuss tomorrow to coordinate this.

vloncar, 2019-01-16:
I made some progress on this. The model is now represented as a graph as opposed it to being a sequential list. This allows the model to have multiple inputs and outputs (needed by #104). Instead of the free-form dicts, the layers are now python objects that can generate their own c++, so I am able to avoid constructing c++ code for the current layer by checking what the previous layer was. Still a lot of work left.

One thing I wanted to discuss is how to handle transformations that hls_writer currently does:

- Sublayers are a bit problematic, and sort of ruin the flow if the layer has to check whether to split or not. I am thinking about handling them not as sublayers, but as regular layers, as if the model had them in the first place (probably in `keras-to-hls`). OTOH, are sublayers about to become obsolete, given the work of Phil and Giuseppe?

- Flatten/Unflatten complicates things a bit as I want to handle them properly (not just as a cpp shim that always preceeds/follows conv layers), but I don't see their usefulness. Conv2d unflattens the input array before calling `conv_2d` function only to flatten it at the beginning of the fucntion. Conv1d doesn't do that, but it is a simple change. Given that we know HLS has issues with multidimensional arrays and we already flatten the weight/bias arrays, we could simplify a lot by working only with flat arrays everywhere.

- Activations can either be an argument of the layer or a layer itself, and handling both cases is a bit messy in current `hls_writer`. It would be a lot simpler if we treated all activations as layers (as some activations are only available as layers). We would still support importing models that use both styles since it is trivial to rewrite activations as layers when parsing them in `keras-to-hls`.

benjaminkreis, 2019-01-16:
Sounds really good!  Graphs are a great idea.

On your points:
1) I do believe sublayers will become obsolete soon, so hopefully you can avoid dealing with this altogether.

2) I think it would be fine to flatten everywhere.  I agree it would make things simpler.  

3) That sounds good.  The only reason for being the way it is now is that originally I did not foresee that activations could be their own layers.

nhanvtran, 2019-01-18:
@vloncar: i dont have much to add on top of what @benjaminkreis said, i agree with what he wrote. 

Graphs make a lot of sense because we 've had this issue where 1 layer is an input to 2 different layers from sergo.

closed 2019-04-04 by benjaminkreis