fastmachinelearning/hls4ml #123: Convert pytorch resnet18 model to HLS

garyhujingyao, 2018-12-04:
Hi all,

I really appreciate your efforts to provide this excellent tool to convert pytorch model to HLS.

I am trying to convert a resnet18 model to HLS, and I found that the example-models inside [pytorch-to-hls folder](https://github.com/hls-fpga-machine-learning/hls4ml/tree/master/pytorch-to-hls/example-models) only convert the linear layers, so I was wondering how I can convert layers such as conv2d, batchnormalization2d, or maxpool2d to the HLS. It would be very helpful if you can give some examples.

Thanks a lot,
Jingyao

nhanvtran, 2018-12-04:
hi @garyhujingyao 

Thanks for your interest!  For PyTorch we do not yet have support for conv layers though we do have some (growing) support in Keras.  

For larger networks such as ResNet18, there is also very active work in scaling up network sizes.  We can keep you posted here on this issue

garyhujingyao, 2018-12-04:
Thanks for your reply @nhanvtran.

I will try myself to convert ResNet18 in Keras, though I think it is not easy :-). Please keep me posted if your guys can scale up network sizes to convert models like ResNets.

mohamedelhewehy, 2022-09-11:
Is this not supproted yet ? 
Also I'm masters student where can I find research areas in hls4ml or using it?

bd0100, 2024-09-27:
Is there any new recent developments in this area?

vloncar, 2024-09-27:
If you're willing to get your hands dirty it's possible. There's proof-of-concept work done on ResNet on multiple FPGAs (see [here](https://indico.cern.ch/event/924283/contributions/4105333/attachments/2154984/3634529/aigean_fastml.pdf)). And if you're willing to do heavy pruning and quantization (to 1 or 2 bits) you might fit into a single chip.

closed 2026-10-28 by github-actions[bot]