fastmachinelearning/hls4ml #124: Errors caused by setting max. number of layers

sherylll, 2018-12-10:
First issue:

when I run `vivado_hls build_prj.tcl` I got the following errors

```
ERROR: [HLS 200-101] 'config_array_partition': Unknown option '-maximum_size'.
ERROR: [HLS 200-101] 'config_array_partition': Unknown option '4096'.
```

Second:

in the code here: https://github.com/hls-fpga-machine-learning/hls4ml/blob/11820f86910d65e2421e2c78ef3dc605ffc2a86b/keras-to-hls/keras-to-hls.py#L188
if the layer is too big and n_subout is zero, the while loop might never end. 
```
MAXMULT = 4096
...
                n_subout = int(MAXMULT/layer['n_in'])
                n_totout = 0
                layer['n_subout'] = []
                layer['weights_n_subzeros'] = []
                layer['n_part'] = 0
                while n_totout < layer['n_out']:
                    if n_totout + n_subout <= layer['n_out']:
                        layer['n_subout'].append(n_subout)
                        n_totout += n_subout                    
                    else:
                        layer['n_subout'].append(layer['n_out']-n_totout)
                        n_totout += layer['n_out']-n_totout
layer['n_part'] += 1
```

vloncar, 2018-12-10:
The first issue is not an actual error (otherwise the script would stop), and you can safely ignore it. It occurs only when using versions of Vivado older than 2018.1. Unfortunately, due to the way `config_array_partition` works, completely suppressing its output is not trivial.

gksks1029, 2026-10-27:
I still see the same messages when using Vitis HLS 2024.2:

--------------------------------------------------------------------------------------
ERROR: [HLS 200-101] config_array_partition: Unknown option '-maximum_size'.
ERROR: [HLS 200-101] config_array_partition: Unknown option '4096'.
--------------------------------------------------------------------------------------

The HLS synthesis and IP export complete successfully despite these messages.

I'd like to investigate whether this compatibility handling can be improved and try contributing a fix. Is there any historical context or compatibility requirement I should keep in mind before changing this part of the Tcl script?

vloncar, 2026-10-27:
Thanks for your interest. As you have observed, it is benign, the synthesis still succeeds. The script tries to set a parameter which no longer exists in newer versions of Vitis HLS. If it fails, it continues since the hard limit doesn't exist in newer versions. It is implemented this way to ensure compatibility with Vivado HLS 2020.1 where setting the maximum (overriding the default 1024) was required. We'll remove it eventually so no message is printed.

closed 2026-10-27 by vloncar