fastmachinelearning/hls4ml #132: Skip connections?

irishjars, 2019-02-18:
Hi,
I have a network that has conv layer that runs parallel to {batchnorm, relu, conv, batchnorm, relu, conv} layers, whose output is finally added to the two using layer.add in Keras. But it looks like HLS4ML does not have support for layer.add.

is there an alternate solution to this?

vloncar, 2019-02-19:
Hi,
Thanks for your interest in hls4ml. Unfortunately, our tool does not support merge layers yet (`Add`, `Substract`, `Concatenate` etc). We plan on adding this once we finish restructuring in #120. Stay tuned.

vinayak19th, 2021-05-11:
@vloncar The concatenate layer support that was added still does not work for 2D convs. Is that a known issue or am I making a mistake with the library?

jmduarte, 2021-09-12:
Merge layers (including skip connections for ResNet-like Conv2D models) are now supported.

closed 2021-09-12 by jmduarte