fastmachinelearning/hls4ml #133: Precision in batch norm

benjaminkreis, 2019-02-23:
@sergojin, @jmduarte, and I found that doing the batch norm computation and casting to `res_T` in one line leads to poorer agreement with floating point than expected.

Replacing this:
```
res[ires] = (res_T) (data[ires]-mean[norm_index])*scale[norm_index]+beta[norm_index]
```

with 
```
ap_fixed<32,14> temp = (data[ires]-mean[ires])*scale[ires]+beta[ires];
res[ires] = (res_T) temp;
```
where ap_fixed<32,14> is some arbitrarily chosen high precision leads to good agreement between `res` and the floating point expectation.  In other words, the casting is not just changing precision at the end of the calculation but is instead influencing the precision of the calculation itself.

It's a small change, but we need to think of something better than just hardcoding a random high precision type for `temp`.

holzman, 2019-02-25:
What types were data_T and res_T in this particular example?

benjaminkreis, 2019-02-26:
IIRC, we were using ap_fixed<18,8> or <16,6>.

sergojin, 2019-02-26:
Correct. <18,8> in that particular case.

holzman, 2019-02-26:
Your original code isn't quite doing what you think it is doing.

```
res[ires] = (res_T) (data[ires]-mean[norm_index])*scale[norm_index]+beta[norm_index]
```

C-style casts have higher precedence than arithmetic operations, so this is casting (data-mean) to res_T but not scale or beta. If you want to cast the whole expression, add another set of parens, or use static_cast.

benjaminkreis, 2019-03-01:
@sergojin, do you remember if we tried adding the parenthesis? Do you have the numbers on hand that showed the problem?  That would make it quick to test.

closed 2019-03-04 by benjaminkreis