fastmachinelearning/hls4ml #143: Example RF Signal Processing Application

ejk43, 2019-05-19:
Hi All,

A cursory investigation into ML for RF signal processing (hls4ml4rf?) based on Jason's recommendations seems pretty promising. It turns out that DeepSig has several public resources that look helpful to get started:

1. Datasets: https://www.deepsig.io/datasets
a. 2018.01A: Live over-the-air data
b. 2016.10a/10b: Simulated datasets, good for initial training

2. Example notebook for a basic Modulation Recognition application: https://github.com/radioML/examples/blob/master/modulation_recognition/RML2016.10a_VTCNN2_example.ipynb

The implemented network is:
- 2D convolution
- 2D convolution
- Dense layer
- Dense layer classifier

Can we recreate the results here, maybe trim some parameters or discretize, and implement with HLS4ML?

Other related ideas worth mentioning but maybe not as promising: 
- Towards the idea of using an "off the shelf" CNN -- If hls4ml can support any of the standard network designs, I expect we could pretty easily try some kind of transfer-learning experiment in the RF domain (or many other fields too, for that matter).
- I have some new ideas about real-world RF event-based "triggering" applications I would love to explore. The end result here would likely be a classifier that indicates whether the event is interesting or not -- sounds like the system may be a similar concept to the the more physics-oriented HLT ML work. This is more of a long term effort since I have no applicable datasets or examples right now.

ejk43, 2019-07-21:
It appears the example "Modulation Recognition" example notebook is a bit outdated  -- it was using python2 and keras pre-2.0. 

I've got a fork now with some updates for python3 and more recent Keras. Of course I'm not totally sure it works with the *most* recent code, since my environment might be getting a little old by now, but I think it's at least slightly healthier. 

I also tested two different network architectures, one with a "dense-only" approach and one with conv2d layers. Both seem to require lots of weights. Any pointers where to start looking for pruning or binary/ternary optimizations in Keras? Might be nice to try a few techniques and compare results?

Duchstf, 2019-09-22:
Hello, just for anyone interested in this issue. I trained a pruned version of the CNN model in the notebook using TensorFlow optimization toolkit. 

The pruned model has a bit higher loss compare with the full model. However, the performance on the test set is about the same. 

The training codes are in this repo: https://github.com/Duchstf/hls4ml-RF/tree/dmh/prune-1
Detailed comparison: https://github.com/Duchstf/hls4ml-RF/blob/dmh/prune-1/eval_performance.ipynb

Here is the output when I compare the h5 zip files:

```
Size of the unpruned model before compression: 5.03 Mb
Size of the unpruned model after compression: 4.35 Mb
Size of the pruned model before compression: 1.69 Mb
Size of the pruned model after compression: 0.33 Mb
```

Duchstf, 2019-09-27:
I used Javier's code for calculating the percentage pruned ... here is the result (the performance is still the same):

Also just a note on the observation that Tensorflow's pruning code apparently always leaves the model with the right "pruning percentage" we put in for `final_sparsity` parameter. 

```
pruning summary
  layer conv1/prune_low_magnitude_conv1/kernel:0: 90.0% weights pruned
  layer conv2/prune_low_magnitude_conv2/kernel:0: 90.0% weights pruned
  layer dense1/prune_low_magnitude_dense1/kernel:0: 90.0% weights pruned
  layer dense2/prune_low_magnitude_dense2/kernel:0: 90.0% weights pruned
model: 90.0% weights pruned
```

Updated notebook with Javier's code:
https://github.com/Duchstf/hls4ml-RF/blob/dmh/prune-1/eval_performance.ipynb

closed 2026-10-28 by github-actions[bot]