fastmachinelearning/hls4ml #146: Package hls4ml with setuptools

vloncar, 2019-06-05:
Create a single user-facing tool and package the project with setuptools so it can be installed with `pip` in the future. More details here:
https://gist.github.com/vloncar/f149c7a0dba34d51e0d9c6d3cb17b12a
(sans the model zoo part for now)

Noah Paladino (@ngpaladi) will work on this.

nhanvtran, 2019-06-06:
https://github.com/vloncar/hls4ml/tree/intel

vloncar, 2019-06-07:
I have created a branch with some initial support for `setuptools`:
https://github.com/vloncar/hls4ml/tree/packaging

Includes a major restructuring of the project packages and introduces a new tool (located in `scripts/hls4ml`) that, when installed, will be available from the command line. A lot of things still need to be done to support running the included `hls4ml` from any location. @nhanvtran @ejk43 comments on the project structure and setuptools are welcome.

Brief example on how to build and install the wheel:
```
python setup.py sdist bdist_wheel
pip install dist/HLS4ML-0.2.0-py3-none-any.whl
```
@ngpaladi If you have any questions, feel free to ask me.

ejk43, 2019-06-08:
Nice! Installation worked for me in a clean conda environment and I could access the hls4ml script on my path. Also, if you're in the hls4ml repo, you can just do `pip install .` which saves a line of typing.

I tried running `hls4ml convert -c <config>` in a out-of-tree test directory but had a few failures, mostly related to typos and not installing the hls-templates package_data. Some minor updates here seemed to fix stuff: https://github.com/ejk43/hls4ml/tree/packaging

I was able to create a project from the example config in a new directory and run through simulation/synthesis! Pretty cool! Very useable.

One odd thing is that the include file for the nnet_utils folder ends up referencing the files inside the conda environment (or wherever hls4ml is installed). Maybe it would make sense to optionally copy the nnet_utils library into some source code directory? Or inside the generated project?

vloncar, 2019-06-08:
Hi @ejk43, thanks for the fixes. I thought that `include_package_data=True` in setup.py should be enough, but the files weren't there.

BTW, is `pip install .` always reinstalling all files of a package? I had issues when using that command to switch between different (major) versions of onnx development branch. Apparently not all files got rebuilt and replaced and that led to many problems, I hope that in our project we can't get into that situation.

Regarding the `nnet_utils`, my preference is to bundle these files with the generated project. That is how I did it for the intel branch. I thought it was a simpler solution as it can save us from users complaining that something broke when they upgrade hls4ml (or conda).

ejk43, 2019-06-08:
> Regarding the nnet_utils, my preference is to bundle these files with the generated project. That is how I did it for the intel branch. I thought it was a simpler solution as it can save us from users complaining that something broke when they upgrade hls4ml (or conda).

Yeah, thats a really good point. I completely agree. My branch refers to the nnet_utils in the conda env but packaging with the generated project is definitely a better approach.

> BTW, is pip install . always reinstalling all files of a package? I had issues when using that command to switch between different (major) versions of onnx development branch. 

Aha, gotcha, pip is sometimes weird. I usually do `pip install -U .` which makes it upgrade the package if you've incremented the version. If that still doesn't work: `pip install -U --force-reinstall .`

vloncar, 2019-06-21:
@ngpaladi I've merged EJ changes and added a fix for `nnet_utils`. You can take over from here.

closed 2019-09-27 by vloncar