fastmachinelearning/hls4ml #147: simulation,validation,speedup

nhanvtran, 2019-06-07:
speeding up simulation and a bunch of scripts for validation scans with many reuse factors

https://indico.cern.ch/event/823156/contributions/3442008/attachments/1850883/3038435/fastml_simulation.pdf

GiuseppeDiGuglielmo, 2019-06-07:
We have a working branch here:
https://github.com/hls-fpga-machine-learning/hls4ml/tree/gdg/validation

I discussed with @vloncar  and he suggested few changes that I totally agree with and I will integrate soon on the branch.

GiuseppeDiGuglielmo, 2019-07-03:
We have a consolidated **validation** flow. The basic idea is to generate two log files with the predictions after both the C simulation (`csim_design`) and the RTL co-simulation (`cosim_design`), respectively. These two files are then compared. Any difference is a symptom of a wrongly generated RTL implementation and must by investigated (maybe a _too aggressive_ HLS configuration, etc.).

There are two implementations of this validation approach:
1. The [C/C++ testbench](https://github.com/hls-fpga-machine-learning/hls4ml/blob/e60d93d12973b7d56f08b1de5958fe55ea56acc6/hls-template/myproject_test.cpp#L74-L101) performs the log comparison (with an OS-dependent `diff` call). I report this for the sake of completeness.
2. The [TCL testbench](https://github.com/hls-fpga-machine-learning/hls4ml/tree/gdg/validation) performs the log comparison. This is neater and is the solution that @vloncar and I agreed on.

Few more notes:
- The first implementation, which embeds the validation in C/C++, is what Xilinx proposes in various guides (for [example](https://www.xilinx.com/support/documentation/sw_manuals/ug1197-vivado-high-level-productivity.pdf) at page 33 - *Self-Checking Test Bench*). The latest versions of the Vivado Suite made commands like `diff` available on Windows too.
- For the second implementation, we were counting on the compilation flag `__RTL_SIMULATION__` to distinguish between C and RTL simulation. This flag should be automatically added by Vivado HLS, but unfortunately, it seems only available for SystemC/RTL co-simulation. @vloncar suggested this [workaround](https://github.com/hls-fpga-machine-learning/hls4ml/blob/gdg/validation/hls-template/build_prj.tcl#L80-L81), but we hope in a future fix from Xilinx. See also this [forum post](https://forums.xilinx.com/t5/Vivado-High-Level-Synthesis-HLS/Distinguish-C-C-sim-vs-RTL-cosim-in-the-testbench/m-p/990948#M17281).

The code on this branch also integrates the feature to **load weights from files on simulation** to avoid long compilation runs and to reduce the overall C-simulation and RTL-co-simulation times. In the `master`-branch implementation, the compilation of the headers which _declare_ and _define_ the weight and bias arrays could take a few hours in the case of large layers. In this branch implementation, we only declare those big arrays in the headers, we [store the weights in TXT files](https://github.com/hls-fpga-machine-learning/hls4ml/blob/gdg/validation/hls-writer/hls_writer.py#L307), and we [load the weights at the beginning of the simulation](https://github.com/hls-fpga-machine-learning/hls4ml/blob/gdg/validation/hls-writer/hls_writer.py#L168-L177). To preserve the HLS flow (we need still need those weights/biases in the headers when ``, we use conditional macros to keep the weights in the headers when we run HLS (`csynth_design`).

At this point, we may be ready for a PR.

closed 2026-10-28 by github-actions[bot]