fastmachinelearning/hls4ml #148: No-Multiplier Optimized Batch Norm

ejk43, 2019-06-15:
Hi all,

I've been considering this idea since we heard the feedback that batch norms are essentially "unnecessary" in FPGA...

The PR for the Optimized Batch norm (https://github.com/hls-fpga-machine-learning/hls4ml/pull/142) restructures the batch norm operation to be scale * data + bias...

It might be feasible to replace the "scale" operation by a multiplier-less version. Either:
1. A power-of-2 scale (would remove multipliers during optimization)
2. Recasting the fixed-point data type, with or without changing the underlying bits (some kind of reinterpret_cast? Not sure what the mechanics are to implement this with ap_fixed, though. Maybe using the ".range()" operation to extract the underlying uint bits?)

A big question here is whether using the "closest power of 2" scale factor is sufficient for the batch norm? Or, if not, what other scale factors could be achievable without using multipliers?

Could be an easy way to get back a few multipliers and/or clock cycles for bigger designs. Thoughts?

thesps, 2019-09-02:
For a Batch norm layer immediately following a dense layer, wouldn't it be possible to fold the Batch norm operations into the dense layer? 
The 'scale * data' would involve using 'scale * weight' for every weight of the dense (so do the product at compile time), and the Batch norm 'bias' could just be added to the real dense layer bias.
It wouldn't work generically for a Batch Norm following the activation, but could work the same way for ReLU, I think.

thesps, 2020-02-14:
The merged Dense + BatchNormalization layer is now in PR #179

closed 2023-04-09 by jmduarte