fastmachinelearning/hls4ml #152: How to create block diagram using IP for Pynq?

dustinjoe, 2019-08-04:
Hello. Thank you for this exciting project!
I am trying to use the ip generated from a Keras model and want to use it in Vivado to generate block diagram for my Pynq-Z1 board. But not quite sure how to connect to the Zynq block. Wondering if I can get some help here?

Another little issue is that I am building a model arch that has more than one Conv1D, I found that if I inserted a dense between two Conv1D layers, the dimension of the second Conv1D cannot be recognized correctly. But only one Conv1D or two consequtive Conv1D layers seem good.

signorgelato, 2019-08-05:
Oh, you can just manually edit the 'keras-to-hls.py' file by changing current_shape at the right places.

dustinjoe, 2019-08-05:
Thanks for your reply. I actually currently just use one conv1d and seems that the model is not that different.

jmduarte, 2021-09-24:
Consider also using the new VivadoAccelerator backend #349, which specifically supports Pynq-Z2/Z1

@nicologhielmetti also has a tutorial which he will publish soon

closed 2023-04-09 by jmduarte