fastmachinelearning/hls4ml #156: Open Source License Conflicts

carl-tercero, 2019-08-12:
Hi, interesting project.  I'm considering using this in a project for which the Apache v2 license isn't an issue.  However, after investigating further, it looks as if hls4ml include GPL v3 licensed code.  As a result, there would be a conflict between the Apache and GPL licenses when hls4ml is incorporated into a project.  Under GPL, the entire project would be considered a derivative work and, therefore, would need to comply with the GPL v3 requirements.

Has this issue been considered?  A possible solution would be to see if EJ Kreinar would re-license his code under Apache v2.

ejk43, 2019-08-12:
Hey guys, I'm open to Apache v2 and I think it's a worthwhile change for hls4ml. However there have been a LOT of other contributors over the last few years so we'd need a more global "signoff" among the developer team. I'd also suggest making a template change too so the code is more broadly attributable instead of just to me (this was more appropriate in the original codebase from two years ago, far less appropriate now) :D But yes I'm happy with a move to Apache v2...

violatingcp, 2019-08-12:
Hi guys

I should check with the lawyers at MIT. I think this is ok but I can
contact them too.

Phil

carl-tercero, 2019-08-13:
Thanks, guys.  I'm happy to help with this in any way needed.

holzman, 2019-08-13:
Folks that have contributed code to hls4ml - in principle to relicense, you'd need signoff from:

Ben Kreis, Dylan Rankin, Javier Duarte, Giuseppe Di Guglielmo, Phil Harris, Jen Ngadiuba, Nhan Tran, Richa Rao, Sergo Jindariani, Sioni Summers, Vladimir Loncar, Zhenbin Wu

vloncar, 2019-08-28:
I propose that we make a "Contributor License Agreement" (CLA) that defines what the hls4ml project can do with the contribution (i.e., the code). We all sign the CLA, and future contributors also sign before their code can be merged. We can use [CLAssistant](https://cla-assistant.io/) to ease this, it integrates nicely with GitHub.

Ideally, we would ask a lawyer to help us with the text of the CLA, but if we don't have one, we can generate one on http://contributoragreements.org/, they look pretty solid.

carl-tercero, 2019-09-06:
Agreed this is probably the right path to take.

Full disclosure: In addition to being an engineer, I'm an IP lawyer.  I'd suggest running this by the lawyers at your organization(s).

carl-tercero, 2019-11-11:
Xilinx's new tools should work for my project.  Closing this issue.  Thanks everyone.

closed 2019-11-11 by carl-tercero