fastmachinelearning/hls4ml #165: KERAS_conv1d build problem

rcetin, 2019-09-18:
Hello,

Thanks for this awesome project! I am trying to migrate my CNN python code to FPGA. Firstly, i wanted to start with ready-to-use designs. So, i have used `KERAS_conv1d.json` file and converted it to HLS using `hls4ml convert -c test-config.yml` command.

Afterwards, i run `vivado_hls -f build_prj.tcl` command to obtain HDL files. This process gave me that error. 
```
ERROR: [XFORM 203-504] Stop unrolling loop 'ConvOut' (firmware/nnet_utils/nnet_conv.h:112) in function 'nnet::conv_1d<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config6>' because it may cause large runtime and excessive memory usage due to increase in code size. Please avoid unrolling the loop or form sub-functions for code in the loop body.
ERROR: [HLS 200-70] Pre-synthesis failed.
command 'ap_source' returned error code
    while executing
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "

INFO: [Common 17-206] Exiting vivado_hls at Wed Sep 18 22:38:48 2019...
```

What can be the problem in here? 
Best regards.

vloncar, 2019-09-19:
Hi @rcetin,

Thank you for your interest in hls4ml project. Unfortunately, the current implementation of CNNs in hls4ml works only for small models, and models like `KERAS_conv1d.json` are too big to handle. You can try the `KERAS_conv1d_small` model. We are currently working on supporting larger convolution layers and hope to have it ready soon, so stay tuned!

Cheers,
Vladimir

rcetin, 2019-09-19:
Hello Vladimir,

Thank you for your fast response. I have tried `KERAS_conv1d_small` model and i got error again in build step.

**OS:** Ubuntu 16.04
**Vivado version:** 2017.4
```
ERROR: [XFORM 203-711] Argument 'res.V' failed dataflow checking: it can only be written in one process function.
WARNING: [XFORM 203-713] Argument 'res.V' has write operations in process function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config10>.0.0.0.0.0.0_Block_.preheader.06_proc' (firmware/nnet_utils/nnet_dense.h:200:21).
WARNING: [XFORM 203-713] Argument 'res.V' has write operations in process function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config10>.0.0.0.0.0.0_Block_.preheader.012_proc' (firmware/nnet_utils/nnet_dense.h:200:21).
WARNING: [XFORM 203-713] Argument 'res.V' has write operations in process function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config10>.0.0.0.0.0.0_Block_.preheader.018_proc' (firmware/nnet_utils/nnet_dense.h:200:21).
WARNING: [XFORM 203-713] Argument 'res.V' has write operations in process function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config10>.0.0.0.0.0.0_Block_.preheader.024_proc' (firmware/nnet_utils/nnet_dense.h:200:21).
WARNING: [XFORM 203-713] Argument 'res.V' has write operations in process function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config10>.0.0.0.0.0.0_Block_.preheader.030_proc' (firmware/nnet_utils/nnet_dense.h:200:21).
ERROR: [HLS 200-70] Pre-synthesis failed.
command 'ap_source' returned error code
    while executing
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "

INFO: [Common 17-206] Exiting vivado_hls at Thu Sep 19 12:27:09 2019...
```

You can also see full log in here:
[https://gist.github.com/rcetin/1e1b98eab585bbf2b4edbba72056536b](url)

Best regards.

vloncar, 2019-09-19:
Can you provide the yaml config file?

rcetin, 2019-09-19:
test-config.yml

```
KerasJson: keras/KERAS_conv1d_small.json
KerasH5:   keras/KERAS_conv1d_small_weights.h5
#InputData: keras/KERAS_3layer_input_features.dat
#OutputPredictions: keras/KERAS_3layer_predictions.dat
OutputDir: my-hls-test
ProjectName: myproject
XilinxPart: xc7z020clg484-1
ClockPeriod: 5

IOType: io_serial # options: io_serial/io_parallel
HLSConfig:
  Model:
    Precision: ap_fixed<16,6>
    ReuseFactor: 1
#  LayerType:
#    Dense:
#      ReuseFactor: 2
#      Strategy: Resource
#      Compression: True
```

vloncar, 2019-09-19:
`io_serial` is buggy, try with `io_parallel`

rcetin, 2019-09-19:
I have tried using `io_parallel`. It is failed in xsim. But in the end, I guess build is completed. The log file is in here.
https://gist.github.com/rcetin/a21cd7f90800bedbe63109eefecb9a9d

Is my build completed successfully?
and, when `io_serial` is used, is there no pipelining?

Regards.

rcetin, 2019-09-20:
I could not paste the full log somehow. I am sorry. Now, you can see the XSIM error in the log.

Regards

vloncar, 2019-09-22:
I have not seen that error before, it looks like it comes from somewhere inside the `xsim` itself, and not related to the generated RTL.

closed 2021-09-12 by jmduarte