fastmachinelearning/hls4ml #170: Batch processing for images

jimd6776, 2019-10-16:
Hi,
Is there any possibility for batch processing of the input images or only single input can be inferred?

Thanks in advance

nhanvtran, 2019-11-04:
There is no explicit batching set up but it's unclear what you mean exactly.  

In principle, you could tile the hls4ml IP several times and batch up processing but that would be a system-level choice.  Or are you talking about using hls4ml within a coprocessor environment? In that case, it would also be something set up at the interface level with the host level API.

jimd6776, 2019-11-07:
I am taking about having a single kernel and giving as input several images so that each image each processed one cycle after another. Can this be done?

nhanvtran, 2019-11-07:
It can be done.  This would be a system-level choice for you, essentially where are you deploying the kernel -- you decide how the data is input.

closed 2019-11-18 by jimd6776