fastmachinelearning/hls4ml #171: activation function problem

Mrdwq, 2019-10-26:
I just wonder whether activation function like tanh or softsign are included in this project and is it possible to add these activation function in. Thanks in advance.

vloncar, 2019-10-26:
Both are supported. Do you have any issues using them?

Mrdwq, 2019-10-26:
I'm trying to use neural network to characterize the behaviour of power amplifier and i've trained a network in tensorflow and found using softsign gained good convergence. Later, i will try to generate a HLS project. Thank you for your reply.

Mrdwq, 2019-10-26:
I installed hls4ml but when i tried to use "hls4ml convert -c keras-config.yml",  it said "bash: hls4ml: command not found". How to solve this problem?

Mrdwq, 2019-10-27:
https://user-images.githubusercontent.com/49830883/67635321-7cc34900-f900-11e9-9a03-144f78e10bd4.png
DSP resource is 100% and synthesis failed.
Here is the keras-config.yml, it's a 60(input_layer)-50(hidden layer)-50(hidden_layer)-2(output layer) neural network
//////
KerasJson: keras_model_files_for_hls_relu/KERAS_2_50neurons_layer_relu.json
KerasH5:   keras_model_files_for_hls_relu/KERAS_2_50neurons_layer_weights_relu.h5
#InputData: keras/KERAS_3layer_input_features.dat
#OutputPredictions: keras/KERAS_3layer_predictions.dat
OutputDir: nn_for_hls_relu
ProjectName: myproject
XilinxPart: xczu9eg-ffvb1156-2-e
ClockPeriod: 5

IOType: io_parallel # options: io_serial/io_parallel
HLSConfig:
  Model:
    Precision: ap_fixed<18,8>
    ReuseFactor: 1
LayerType:
  Dense:
    ReuseFactor: 2
    Strategy: Resource
    Compression: True
//////

is there any way to optimize or just hardware limited?

vloncar, 2019-10-27:
Increase the reuse factor and switch to resource strategy for the model.

jmduarte, 2021-09-12:
Closed due to inactivity.

closed 2021-09-12 by jmduarte