fastmachinelearning/hls4ml #172: Support "reshape" layer?

Duchstf, 2019-11-07:
Hello,

I'm trying to synthesize a CNN model which uses Keras's reshape in its input layer, but it's currently not supported. 

I think I can work around this by manually reshape the data, but is this something that hls4ml is planning on supporting in the future? If the reshape layer is used in the middle of the network, then there is no easy workaround for this. 

Thanks.

vloncar, 2019-11-07:
There is a solution for this implemented in hack7 branch. You can find the
branch in my fork.

Duchstf, 2019-11-08:
I got pass the reshape error, but this is the new error ...

```
Loading configuration from keras-config.yml
Interpreting Sequential
Input shape: [2, 128]
Topology:
Traceback (most recent call last):
  File "/home/dhoang/miniconda3/bin/hls4ml", line 124, in <module>
    main()
  File "/home/dhoang/miniconda3/bin/hls4ml", line 58, in main
    args.func(args)
  File "/home/dhoang/miniconda3/bin/hls4ml", line 73, in convert
    model = hls4ml.converters.keras_to_hls(yamlConfig)
  File "/home/dhoang/miniconda3/lib/python3.7/site-packages/hls4ml/converters/keras_to_hls.py", line 184, in keras_to_hls
    weights_shape = get_weights_shape(yamlConfig['KerasH5'], layer['name'])
  File "/home/dhoang/miniconda3/lib/python3.7/site-packages/hls4ml/converters/keras_to_hls.py", line 36, in get_weights_shape
    found_data = h5file[layer_name].visit(h5_visitor_func)
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "/home/dhoang/miniconda3/lib/python3.7/site-packages/h5py/_hl/group.py", line 264, in __getitem__
    oid = h5o.open(self.id, self._e(name), lapl=self._lapl)
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py/h5o.pyx", line 190, in h5py.h5o.open
KeyError: "Unable to open object (object 'conv1' doesn't exist)"
```

vloncar, 2019-11-08:
Two possible causes:
1) layer name gets mangled (to be c++ compatible)
2) the format of the h5 changed. Look for string /{}/{} and change to {}/{}
sorry I don't know the line, I'm not near a computer

Duchstf, 2019-11-08:
Oh I just have to change the way I saved the h5 file. This is the new error though ... 

```
Loading configuration from keras-config.yml
Interpreting Sequential
Input shape: [2, 128]
Topology:
Layer name: conv1, layer type: Conv2D, current shape: [None, 1, 121, 64]
Layer name: conv2, layer type: Conv2D, current shape: [None, 1, 90, 32]
Layer name: dense1, layer type: Dense, current shape: [None, 128]
Layer name: dense2, layer type: Dense, current shape: [None, 11]
Layer name: activation, layer type: Activation, current shape: [None, 11]
Creating HLS model
Traceback (most recent call last):
  File "/home/dhoang/miniconda3/bin/hls4ml", line 124, in <module>
    main()
  File "/home/dhoang/miniconda3/bin/hls4ml", line 58, in main
    args.func(args)
  File "/home/dhoang/miniconda3/bin/hls4ml", line 73, in convert
    model = hls4ml.converters.keras_to_hls(yamlConfig)
  File "/home/dhoang/miniconda3/lib/python3.7/site-packages/hls4ml/converters/keras_to_hls.py", line 364, in keras_to_hls
    hls_model = HLSModel(yamlConfig, reader, layer_list, input_layers, output_layers)
  File "/home/dhoang/miniconda3/lib/python3.7/site-packages/hls4ml/model/hls_model.py", line 195, in __init__
    self.config = HLSConfig(config)
  File "/home/dhoang/miniconda3/lib/python3.7/site-packages/hls4ml/model/hls_model.py", line 33, in __init__
    self._parse_hls_config()
  File "/home/dhoang/miniconda3/lib/python3.7/site-packages/hls4ml/model/hls_model.py", line 105, in _parse_hls_config
    hls_config = self.config['HLSConfig']
KeyError: 'HLSConfig'
```

vloncar, 2019-11-08:
Do you have that key in config file? Should be there. If that is not the
cause then I'll need some time to look into it.

Duchstf, 2019-11-08:
Oh I was using the config file from the documentation, which is not updated since hack6 get merged. I'll update it right now. 

But there is still another error: 

```
Loading configuration from keras-config.yml
Interpreting Sequential
Input shape: [2, 128]
Topology:
Layer name: conv1, layer type: Conv2D, current shape: [None, 1, 121, 64]
Layer name: conv2, layer type: Conv2D, current shape: [None, 1, 90, 32]
Layer name: dense1, layer type: Dense, current shape: [None, 128]
Layer name: dense2, layer type: Dense, current shape: [None, 11]
Layer name: activation, layer type: Activation, current shape: [None, 11]
Creating HLS model
Writing HLS project
Traceback (most recent call last):
  File "/home/dhoang/miniconda3/bin/hls4ml", line 124, in <module>
    main()
  File "/home/dhoang/miniconda3/bin/hls4ml", line 58, in main
    args.func(args)
  File "/home/dhoang/miniconda3/bin/hls4ml", line 76, in convert
    model.config.writer.write_hls(model)
  File "/home/dhoang/miniconda3/lib/python3.7/site-packages/hls4ml/writer/vivado_writer.py", line 387, in write_hls
    self.write_parameters(model)
  File "/home/dhoang/miniconda3/lib/python3.7/site-packages/hls4ml/writer/vivado_writer.py", line 218, in write_parameters
    config = layer.config_cpp()
  File "/home/dhoang/miniconda3/lib/python3.7/site-packages/hls4ml/model/hls_model.py", line 813, in config_cpp
    params['n_chan'] = self.get_input_variable().dim_names[2]
IndexError: list index out of range
```

vloncar, 2019-11-09:
Looks like a parsing issue in the input layer. Only two dimensions are read, so `dim_names[2]` fails. Can you send me the model json and h5?

vloncar, 2019-11-12:
I've improved the way Reshape layer is handled (check the latest change in hack7 branch), it now supports changing the number of dimensions, which is what triggered the bug. In your case however, it seems that you are using the Reshape layer in a wrong way and that you really don't need it since you can simply change the shape of the input layer.

Duchstf, 2019-11-14:
Fixed it, converted successfully, thank you!

closed 2019-11-14 by Duchstf