fastmachinelearning/hls4ml #173: the particular FPGA part number?

mk123qwe, 2019-11-11:
Thank you for your great work！
What kind of FPGA should I consider to run a simple example.
Xilinx Virtex-7 is too expensive.

nhanvtran, 2019-11-17:
@mk123qwe to run a simple example, you can choose whichever device you want, but in general, there are a large range of devices you could use

Sejudyblues, 2019-12-12:
> @ mk123qwe运行一个简单的示例，您可以选择所需的任何设备，但是通常，您可以使用很多设备

I am ready to use KC705. i dont know if it can running a  demo.

essa2307, 2019-12-15:
If I change the Xilinx Part number in the yml, it gives me KeyError: 'XilinxPart' Could you please tell me where can I find a list of valid parts?

nhanvtran, 2019-12-17:
Please have a look at the datasheets.  As an example:
https://www.xilinx.com/support/documentation/data_sheets/ds890-ultrascale-overview.pdf

See page 44 for a detailed explanation of the part numbers

closed 2020-09-23 by jmduarte