fastmachinelearning/hls4ml #174: Errors when converting pytorch model

Duchstf, 2019-11-26:
I got this when testing `three_layer_model.pt`. The torch version I'm using is 1.3.1. Maybe downgrading torch would help? 

```
Loading configuration from pytorch-config.yml
Interpreting Model
/home/dhoang/miniconda3/lib/python3.7/site-packages/torch/serialization.py:493: SourceChangeWarning: source code of class 'torch.nn.modules.container.Sequential' has changed. you can retrieve the original source code by accessing the object's source attribute or set `torch.nn.Module.dump_patches = True` and use the patch tool to revert the changes.
  warnings.warn(msg, SourceChangeWarning)
/home/dhoang/miniconda3/lib/python3.7/site-packages/torch/serialization.py:493: SourceChangeWarning: source code of class 'torch.nn.modules.linear.Linear' has changed. you can retrieve the original source code by accessing the object's source attribute or set `torch.nn.Module.dump_patches = True` and use the patch tool to revert the changes.
  warnings.warn(msg, SourceChangeWarning)
/home/dhoang/miniconda3/lib/python3.7/site-packages/torch/serialization.py:493: SourceChangeWarning: source code of class 'torch.nn.modules.activation.ReLU' has changed. you can retrieve the original source code by accessing the object's source attribute or set `torch.nn.Module.dump_patches = True` and use the patch tool to revert the changes.
  warnings.warn(msg, SourceChangeWarning)
/home/dhoang/miniconda3/lib/python3.7/site-packages/torch/serialization.py:493: SourceChangeWarning: source code of class 'torch.nn.modules.activation.Softmax' has changed. you can retrieve the original source code by accessing the object's source attribute or set `torch.nn.Module.dump_patches = True` and use the patch tool to revert the changes.
  warnings.warn(msg, SourceChangeWarning)
Traceback (most recent call last):
  File "/home/dhoang/miniconda3/bin/hls4ml", line 93, in <module>
    main()
  File "/home/dhoang/miniconda3/bin/hls4ml", line 46, in main
    args.func(args)
  File "/home/dhoang/miniconda3/bin/hls4ml", line 59, in convert
    model = hls4ml.converters.pytorch_to_hls(yamlConfig)
  File "/home/dhoang/miniconda3/lib/python3.7/site-packages/hls4ml/converters/pytorch_to_hls.py", line 43, in pytorch_to_hls
    reader = PyTorchDataReader(yamlConfig)
  File "/home/dhoang/miniconda3/lib/python3.7/site-packages/hls4ml/converters/pytorch_to_hls.py", line 25, in __init__
    self.state_dict = self.torch_model.state_dict()
  File "/home/dhoang/miniconda3/lib/python3.7/site-packages/torch/nn/modules/module.py", line 695, in state_dict
    module.state_dict(destination, prefix + name + '.', keep_vars=keep_vars)
  File "/home/dhoang/miniconda3/lib/python3.7/site-packages/torch/nn/modules/module.py", line 696, in state_dict
    for hook in self._state_dict_hooks.values():
  File "/home/dhoang/miniconda3/lib/python3.7/site-packages/torch/nn/modules/module.py", line 585, in __getattr__
    type(self).__name__, name))
AttributeError: 'Softmax' object has no attribute '_state_dict_hooks'
```

vloncar, 2019-11-26:
I think this is due to the .pt file being created with the old PyTorch, so newer versions cannot load it. The model is simple, you can create a new one and train it. Look into [pytorch-training](https://github.com/hls-fpga-machine-learning/pytorch-training) project.

Duchstf, 2019-12-24:
Thank you! This was indeed the issue. I updated the training code in [pytorch-training](https://github.com/hls-fpga-machine-learning/pytorch-training) and created new Pytorch example models, which are documented [here](https://github.com/hls-fpga-machine-learning/models/tree/master/pytorch).

closed 2019-12-24 by Duchstf