fastmachinelearning/hls4ml #176: Support for additional layers

Neelam96, 2019-12-30:
Does your tool provides support for PReLU, max pool and soft max.

Duchstf, 2019-12-30:
I think PReLU, maxpooling1D and maxpooling 2D are supported for Keras. I'm not sure for softmax though. I documented the supported layers for Keras [here](https://github.com/hls-fpga-machine-learning/models/tree/master/keras). (This was updated a while ago, not sure if other people want to add anything?)

Neelam96, 2019-12-31:
How I can add Softmax to your tool. Can u please tell me the procedure if there is one.
Thanks

zhenbinwu, 2019-12-31:
We have a version of Softmax in [HLS](https://github.com/hls-fpga-machine-learning/hls4ml/blob/master/hls4ml/templates/vivado/nnet_utils/nnet_activation.h#L212). Might need to add the interface of Keras.

jmduarte, 2021-09-12:
PreLU, Softmax, and MaxPooling1D/2D are all supported for Keras (and potentially PyTorch/ONNX as well).

https://github.com/abdelabd/hls4ml/blob/pyg_to_hls_rebase/hls4ml/model/hls_layers.py#L1358
https://github.com/abdelabd/hls4ml/blob/pyg_to_hls_rebase/hls4ml/model/hls_layers.py#L1371
https://github.com/abdelabd/hls4ml/blob/pyg_to_hls_rebase/hls4ml/model/hls_layers.py#L1147
https://github.com/abdelabd/hls4ml/blob/pyg_to_hls_rebase/hls4ml/model/hls_layers.py#L1178

closed 2021-09-12 by jmduarte