fastmachinelearning/hls4ml #177: Large Keras model does not synthesize on master

Duchstf, 2020-01-03:
Hello,

Just a note about `KERAS_dense_16x200x200x200x200x200x5.json` or bigger models not being able to synthesize using hls4ml from `master` branch (or `hack7` branch from Vladimir for that matters). I tried various RFs (up to 40k) but it does not work. 

I installed an older version of hls4ml from branch [nvt/large_mlp_wDep_v6](https://github.com/hls-fpga-machine-learning/hls4ml/tree/nvt/large_mlp_wDep_v6). This successfully synthesizes the mentioned model. 

Just wondering whether something might get lost when the big pull request for packaging get merged.

harnhua, 2020-01-22:
A bit off-topic, but have switched to this older branch from the current master, the Python packaging scripts seem to be gone now and "pip install ." doesn't work now.

How did you install hls4ml using this nvt/large_mlp_wDep_v6?
(I'm not familiar with conda/Python yet)

vloncar, 2020-01-22:
@harnhua there are install scripts in the root of the project. large_mlp was an experimental branch based on very old version of hls4ml and may contain other issues.

harnhua, 2020-01-22:
@vloncar Thank you for the note on this branch. It sounds like I should not use it instead.
Switching tack, how then can I help resolve the synthesis problems with the large Keras models? (If you are the right person to ask)

vloncar, 2020-01-22:
I'm investigating the issue, it seems to be related to Vivado versions, since the model mentioned synthesizes in Vivado 2019.1, but not in 2018.2. It does take 5x more time. The issue seems related to the optimized path for RF<N_IN. [Here](https://gist.github.com/vloncar/96765d8a71bcac6dfd6a4e8c0ba80eb1) is the version with that stripped out, it does synthesize faster (n 2019.1), however it uses a lot more resources.

Duchstf, 2020-01-22:
@harnhua Just a quick note. If you are willing to try out the old branch, here is a tutorial on how to use the old version of `hls4ml`: 

https://github.com/FPGA4HEP/course_material/blob/paris/part1_hls4ml_intro.md

The configuration file is also a little bit different, but that shouldn't be too hard to figure out.

harnhua, 2020-01-24:
Thank you very much @vloncar 
Do you have a version of nnet_dense.h with the offending portion stripped out as well?
I replaced nnet_dense_large.h with the version that you provided but realized that the error came from nnet_dense.h instead.
```
INFO: [HLS 200-489] Unrolling loop 'Product1' (firmware/nnet_utils/nnet_dense.h:160) in function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config10>' completely with a factor of 200.
ERROR: [XFORM 203-504] Stop unrolling loop 'Product1' (firmware/nnet_utils/nnet_dense.h:160) in function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config10>' because it may cause large runtime and excessive memory usage due to increase in code size. Please avoid unrolling the loop or form sub-functions for code in the loop body.
ERROR: [HLS 200-70] Pre-synthesis failed.
```

harnhua, 2020-01-24:
Thank you very much @Duchstf !
I'll definitely try this out.

vloncar, 2020-01-24:
@harnhua If your layer is large, you should use the `Strategy: Resource` in your yml file. This will switch to the `dense_large` algorithm. See the [configuration](https://fastmachinelearning.org/hls4ml/setup/CONFIGURATION.html) part of the docs.

harnhua, 2020-01-30:
Here are the results of an attempt using "Strategy: Resource" on the dense 16x100... model.

- I tried the 16x100... model this time
- It seems to get further but eventually got "ERROR: [XFORM 203-103] Array 'mult.V' (firmware/nnet_utils/nnet_dense.h:114): partitioned elements number (10000) has exeeded the threshold (4096), which may cause long run-time.
ERROR: [HLS 200-70] Pre-synthesis failed.
"
- Attaching the full log in case it helps.
[keras_dense_16x100x100x100x100x100x5_resource.log](https://github.com/hls-fpga-machine-learning/hls4ml/files/4132055/keras_dense_16x100x100x100x100x100x5_resource.log)

Here is my keras-config.yml:
```
KerasJson: keras/KERAS_dense_16x100x100x100x100x100x5.json
KerasH5:   keras/KERAS_dense_16x100x100x100x100x100x5_weights.h5
#InputData: keras/KERAS_3layer_input_features.dat
#OutputPredictions: keras/KERAS_3layer_predictions.dat
OutputDir: keras_dense_16x100x100x100x100x100x5
ProjectName: myproject
XilinxPart: xcku115-flvb2104-2-i
ClockPeriod: 5

IOType: io_parallel # options: io_serial/io_parallel
HLSConfig:
  Model:
    Precision: ap_fixed<16,6>
    ReuseFactor: 1
  Strategy: Resource
#  LayerType:
#    Dense:
#      ReuseFactor: 2
#      Strategy: Resource
#      Compression: True

```
[keras_dense_16x100x100x100x100x100x5_resource.log](https://github.com/hls-fpga-machine-learning/hls4ml/files/4132061/keras_dense_16x100x100x100x100x100x5_resource.log)

Duchstf, 2020-01-30:
@harnhua Hi, this model should work just fine with the master branch. From your configuration file it seems like you need to have a larger reuse factor (if you use reuse factor 1 with resource strategy, it doesn't do anything different than latency strategy). 

I actually documented the output log file results and configuration file for synthesizing this model in [this repo.](https://github.com/hls-fpga-machine-learning/models/tree/master/keras/KERAS_dense)

Hopefully that helps!

Duc.

closed 2026-10-27 by github-actions[bot]