fastmachinelearning/hls4ml #181: Unable to run example model in Vivado hls2018.3

pentapatisivani, 2020-03-04:
I tried running the example project by adding all files in nnet_utils in my project but still I am getting this error of datatype mismatch while running synthesis even build.tcl isnt working. How exactly can we include hls4ml in vivado hls?
ERROR: [HLS 200-70] Compilation errors found: In file included from Gnet/files/myproject.cpp:1:
In file included from Gnet/files/myproject.cpp:21:
In file included from Gnet/files/parameters.h:7:
In file included from Gnet/files/nnet_dense.h:24:
Gnet/files/nnet_helpers.h:177:1: error: unknown type name 'constexpr'
constexpr int ceillog2(int x){
^
Gnet/files/nnet_helpers.h:177:11: error: expected unqualified-id
constexpr int ceillog2(int x){
          ^
Gnet/files/nnet_helpers.h:181:1: error: unknown type name 'constexpr'
constexpr int pow2(int x){
^
Gnet/files/nnet_helpers.h:181:11: error: expected unqualified-id
constexpr int pow2(int x){
          ^
In file included from Gnet/files/myproject.cpp:1:
In file included from Gnet/files/myproject.cpp:21:
In file included from Gnet/files/parameters.h:7:
Gnet/files/nnet_dense.h:56:22: error: no template named 'enable_if' in namespace 'std'; did you mean '_ap_type::enable_if'?
inline typename std::enable_if<std::is_same<data_T, ap_uint<1>>::value
                ~~~~~^~~~~~~~~
                     _ap_type::enable_if
F:/apps/Vivado/2018.3/common/technology/autopilot\ap_common.h:190:8: note: '_ap_type::enable_if' declared here
struct enable_if {};
       ^
In file included from Gnet/files/myproject.cpp:1:
In file included from Gnet/files/myproject.cpp:21:
In file included from Gnet/files/parameters.h:7:
Gnet/files/nnet_dense.h:56:37: error: no member named 'is_same' in namespace 'std'
inline typename std::enable_if<std::is_same<data_T, ap_uint<1>>::value
                               ~~~~~^
Gnet/files/nnet_dense.h:56:45: error: 'data_T' does not refer to a value
inline typename std::enable_if<std::is_same<data_T, ap_uint<1>>::value
                                            ^
Gnet/files/nnet_dense.h:55:16: note: declared here
template<class data_T, class weight_T, class ret_T>
               ^
Gnet/files/nnet_dense.h:65:1: error: expected unqualified-id
inline typename std::enable_if<(not std::is_same<data_T, ap_uint<1>>::value)
^
In file included from Gnet/files/myproject.cpp:1:
Gnet/files/myproject.cpp:66:8: error: no member named 'conv_2d' in namespace 'nnet'
 nnet::conv_2d<input_t, result_t, config1>(data, layer1_out, w1, b1);
 ~~~~~~^
Gnet/files/myproject.cpp:66:16: error: unexpected type name 'input_t': expected expression
 nnet::conv_2d<input_t, result_t, config1>(data, layer1_out, w1, b1);
               ^
Gnet/files/myproject.cpp:66:25: error: unexpected type name 'result_t': expected expression
 nnet::conv_2d<input_t, result_t, config1>(data, layer1_out, w1, b1);
                        ^
Gnet/files/myproject.cpp:66:35: error: 'config1' does not refer to a value
 nnet::conv_2d<input_t, result_t, config1>(data, layer1_out, w1, b1);
                                  ^
Gnet/files/parameters.h:39:8: note: declared here
struct config1 : nnet::conv2d_config {
       ^
In file included from Gnet/files/myproject.cpp:1:
Gnet/files/myproject.cpp:66:44: warning: expression result unused [-Wunused-value]
 nnet::conv_2d<input_t, result_t, config1>(data, layer1_out, w1, b1);
                                           ^~~~
Gnet/files/myproject.cpp:66:50: warning: expression result unused [-Wunused-value]
 nnet::conv_2d<input_t, result_t, config1>(data, layer1_out, w1, b1);
                                                 ^~~~~~~~~~
Gnet/files/myproject.cpp:66:62: warning: expression result unused [-Wunused-value]
 nnet::conv_2d<input_t, result_t, config1>(data, layer1_out, w1, b1);
                                                             ^~
Gnet/files/myproject.cpp:81:8: error: no member named 'dense' in namespace 'nnet'
 nnet::dense<input_t, result_t, config2>(layer1_out_flat_relu, logits2, w2, b2);
[... 51 lines cut ...]
F:/apps/Vivado/2018.3/common/technology/autopilot\ap_fixed_base.h:868:69: note: in instantiation of member function 'ap_fixed_base<51, 41, true, 5, 3, 0>::to_ap_int_base' requested here
  inline __attribute__((always_inline)) int to_int() const { return to_ap_int_base().to_int(); }
                                                                    ^
F:/apps/Vivado/2018.3/common/technology/autopilot\ap_fixed_base.h:1041:71: note: in instantiation of member function 'ap_fixed_base<51, 41, true, 5, 3, 0>::to_int' requested here
  inline __attribute__((always_inline)) operator int() const { return to_int(); }
                                                                      ^
Gnet/files/nnet_activation.h:264:27: note: in instantiation of member function 'ap_fixed_base<51, 41, true, 5, 3, 0>::operator int' requested here
      int exp_res_index = exp_res[ii]*CONFIG_T::table_size/64;
                          ^
Gnet/files/myproject.cpp:84:5: note: in instantiation of function template specialization 'nnet::softmax<ap_fixed<18, 8, 5, 3, 0>, ap_fixed<18, 8, 5, 3, 0>, softmax_config2>' requested here
    nnet::softmax<result_t, result_t, softmax_config2>(logits2, res);
    ^
8 warnings and 16 errors generated.
Failed during preprocessing.
    while executing
"source C:/Users/dell/Desktop/conv/solution1/csynth.tcl"
    invoked from within
"hls::main C:/Users/dell/Desktop/conv/solution1/csynth.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
"hls_proc $argv"
Finished C synthesis.

Duchstf, 2020-03-04:
Hi, what specific example project are you running? Did you try following the workflow in our [documentation](https://fastmachinelearning.org/hls4ml/setup/QUICKSTART.html)? Or did you just copy the files over to your project?

Also, I think the example projects in master branch are quite outdated, can you try again with the examples in this [repo](https://github.com/hls-fpga-machine-learning/models)? 

Duc.

thesps, 2020-03-04:
Are you integrating the NN into a larger project or just compiling the NN with hls4ml?
From your error, it looks like you don't have the C++0x flag. You can see how it's used in the hls4ml build_prj.tcl file, for example:
`add_files firmware/myproject.cpp -cflags "-std=c++0x"`
If you're integrating the code into another project, you should add the `-cflags "-std=c++0x"` flag to your build script. Otherwise, simply use `hls4ml build -p your_project -s` which will use the build script we generate.

pentapatisivani, 2020-03-04:
> Hi, what specific example project are you running? Did you try following the workflow in our [documentation](https://fastmachinelearning.org/hls4ml/setup/QUICKSTART.html)? Or did you just copy the [...]

Thank you for mentioning about the updated repo. I ran the new tcl script directly in hls terminal to create the project I have got errors related to pre-existing files in vivado hls 
[image]
. How can I resolve this issue? 
(Attaching screenshot of result)

***** C SIMULATION COMPLETED IN 0h0m41s *****                                                                                                                                                                                                ***** C/RTL SYNTHESIS *****                                                                                                                                                                                                                  INFO: [SCHED 204-61] Option 'relax_ii_for_timing' is enabled, will increase II to preserve clock frequency constraints.                                                                                                                      INFO: [HLS 200-10] Analyzing design file 'firmware/myproject.cpp' ...                                                                                                                                                                        ERROR: [HLS 200-70] Compilation errors found: In file included from firmware/myproject.cpp:1:                                                                                                                                                In file included from firmware/myproject.cpp:19:                                                                                                                                                                                             In file included from F:/apps/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\iostream:39:                                                                                                                   In file included from F:/apps/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\ostream:39:                                                                                                                    In file included from F:/apps/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\ios:39:                                                                                                                        In file included from F:/apps/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\exception:151:                                                                                                                 F:/apps/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\exception_ptr.h:132:13: error: unknown type name 'type_info'                                                                                               const type_info*                                                                                                                                                                                                                                   ^                                                                                                                                                                                                                                In file included from firmware/myproject.cpp:1:                                                                                                                                                                                              In file included from firmware/myproject.cpp:19:                                                                                                                                                                                             In file included from F:/apps/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\iostream:39:                                                                                                                   In file included from F:/apps/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\ostream:39:                                                                                                                    In file included from F:/apps/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\ios:39:                                                                                                                        In file included from F:/apps/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\exception:151:                                                                                                                 F:/apps/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:62:5: error: the parameter for this explicitly-defaulted copy constructor is const, but a member or base requires it to be non-const                                                                                                                                                                                                                                               nested_exception(const nested_exception&) = default;   [... cut ...]                                                                                                                                                                                                                                                                                                                                                                                                                                              INFO: [Common 17-206] Exiting vivado_hls at Wed Mar  4 21:49:44 2020...

vloncar, 2020-03-04:
There is a workaround [here](https://github.com/hls-fpga-machine-learning/hls4ml/pull/131#issuecomment-467811547), however the side-effects of doing this are unknown since most of us don't use Windows for development.

If you have a contact within Xilinx, I would suggest also complaining about this to them. They ship a standard library that was part of a compiler from 2010, which is no longer funny, now it's just sad.

pentapatisivani, 2020-03-08:
Thanks a lot! That fixed the issue and is there any reference or documentation for hls4ml to know what exactly each function is doing? For example there are two functions for conv2d conv2d_latency_cl and conv2d_latency_cf (In this repo: https://github.com/hls-fpga-machine-learning/models ). What exactly is the difference between these 2? and nnet_utils itself has many files and functions and its being difficult to understand what function does what just from the codes and there are multiple functions with similar code. Can you provide any resource with proper documentation of code. And the example models only include 4-5 functions

thesps, 2020-03-09:
For the specific case `conv2d_latency_cl` vs `conv2d_latency_cf` the `cl` means 'Channels Last' while `cf` means 'Channels First'. The function which is used will be determined by the configuration of the trained model you provide.

Usually the function which gets used is controlled either by your model, as above, or your configuration file. For example for the Dense layer we have `Strategy: Latency`, `Strategy: Resource`, `Compression: {True/False}`. These, together with the `Reuse` factor will define which function gets used to give the best balance of FPGA resources, latency and synthesizability. If you want to explore the design space it's better to try out these options in the config than to modify which function is called in the produced C++.

The best documentation for the config file parameters is https://fastmachinelearning.org/hls4ml/setup/CONFIGURATION.html

closed 2020-03-09 by thesps

rtspk, 2021-12-04:
@pentapatisivani  Heyy, How did you solve your problem ? it seems i am facing the same problem. 

When i executed this line  ` hls_model.compile()` , it did create an HLS project. but hls model compilation didnt complete. 
( I am using linux , and setup the conda enviroment using `conda env create -f environment.yml ` on hls_tutorials page)

**Console Output:**

Writing HLS project
Done
In file included from firmware/parameters.h:7:0,
                 from firmware/conv2d_1L_1CH_2x2K.cpp:22:
firmware/nnet_utils/nnet_helpers.h: In function ‘void nnet::load_compressed_weights_from_txt(T*, const char*)’:
firmware/nnet_utils/nnet_helpers.h:92:26: error: range-based ‘for’ loops are not allowed in C++98 mode
             for (char c: extra_chars) {
                          ^
firmware/nnet_utils/nnet_helpers.h:95:23: error: ‘std::string’ has no member named ‘back’
             if (token.back() == ',') {
                       ^
firmware/nnet_utils/nnet_helpers.h: In function ‘void nnet::load_exponent_weights_from_txt(T*, const char*)’:
firmware/nnet_utils/nnet_helpers.h:138:26: error: range-based ‘for’ loops are not allowed in C++98 mode
             for (char c: extra_chars) {
                          ^
firmware/nnet_utils/nnet_helpers.h:141:23: error: ‘std::string’ has no member named ‘back’
             if (token.back() == ',') {
                       ^
firmware/nnet_utils/nnet_helpers.h: In function ‘void nnet::save_layer_output(data_T*, const char*, size_t)’:
firmware/nnet_utils/nnet_helpers.h:233:47: error: no matching function for call to ‘std::basic_fstream<char>::open(std::basic_ostringstream<char>::__string_type&, const openmode&)’
         out.open(filename.str(), std::ios::app);
                                               ^
firmware/nnet_utils/nnet_helpers.h:233:47: note: candidate is:
In file included from firmware/nnet_utils/nnet_helpers.h:26:0,
                 from firmware/parameters.h:7,
                 from firmware/conv2d_1L_1CH_2x2K.cpp:22:
/usr/include/c++/4.8.2/fstream:886:7: note: void std::basic_fstream<_CharT, _Traits>::open(const char*, std::ios_base::openmode) [with _CharT = char; _Traits = std::char_traits<char>; std::ios_base::openmode = std::_Ios_Openmode]
       open(const char* __s,
       ^
/usr/include/c++/4.8.2/fstream:886:7: note:   no known conversion for argument 1 from ‘std::basic_ostringstream<char>::__string_type {aka std::basic_string<char>}’ to ‘const char*’
In file included from firmware/parameters.h:7:0,
                 from firmware/conv2d_1L_1CH_2x2K.cpp:22:
firmware/nnet_utils/nnet_helpers.h: In function ‘void nnet::save_layer_output(hls::stream<srcType>&, const char*, size_t)’:
firmware/nnet_utils/nnet_helpers.h:263:47: error: no matching function for call to ‘std::basic_fstream<char>::open(std::basic_ostringstream<char>::__string_type&, const openmode&)’
         out.open(filename.str(), std::ios::app);
                                               ^
firmware/nnet_utils/nnet_helpers.h:263:47: note: candidate is:
In file included from firmware/nnet_utils/nnet_helpers.h:26:0,
                 from firmware/parameters.h:7,
                 from firmware/conv2d_1L_1CH_2x2K.cpp:22:
/usr/include/c++/4.8.2/fstream:886:7: note: void std::basic_fstream<_CharT, _Traits>::open(const char*, std::ios_base::openmode) [with _CharT = char; _Traits = std::char_traits<char>; std::ios_base::openmode = std::_Ios_Openmode]
       open(const char* __s,
       ^
/usr/include/c++/4.8.2/fstream:886:7: note:   no known conversion for argument 1 from ‘std::basic_ostringstream<char>::__string_type {aka std::basic_string<char>}’ to ‘const char*’
In file included from firmware/parameters.h:7:0,
                 from firmware/conv2d_1L_1CH_2x2K.cpp:22:
firmware/nnet_utils/nnet_helpers.h: In function ‘void nnet::copy_data_axi(std::vector<_Tp>, dst_T*)’:
firmware/nnet_utils/nnet_helpers.h:305:14: error: ‘i’ does not name a type
     for(auto i = 0; i < SIZE; i++)
              ^
firmware/nnet_utils/nnet_helpers.h:305:21: error: expected ‘;’ before ‘i’
     for(auto i = 0; i < SIZE; i++)
                     ^
firmware/nnet_utils/nnet_helpers.h:305:21: error: ‘i’ was not declared in this scope
firmware/nnet_utils/nnet_helpers.h: At global scope:
firmware/nnet_utils/nnet_helpers.h:419:1: error: ‘constexpr’ does not name a type
 constexpr int ceillog2(int x){
 ^
firmware/nnet_utils/nnet_helpers.h:419:1: note: C++11 ‘constexpr’ only available with -std=c++11 or -std=gnu++11
firmware/nnet_utils/nnet_helpers.h:423:1: error: ‘constexpr’ does not name a type
 constexpr int floorlog2(int x){
 ^
firmware/nnet_utils/nnet_helpers.h:423:1: note: C++11 ‘constexpr’ only available with -std=c++11 or -std=gnu++11
[... 340 lines cut ...]
 ^
firmware/nnet_utils/nnet_helpers.h:423:1: note: C++11 ‘constexpr’ only available with -std=c++11 or -std=gnu++11
firmware/nnet_utils/nnet_helpers.h:427:1: error: ‘constexpr’ does not name a type
 constexpr int pow2(int x){
 ^
firmware/nnet_utils/nnet_helpers.h:427:1: note: C++11 ‘constexpr’ only available with -std=c++11 or -std=gnu++11
g++: error: conv2d_1L_1CH_2x2K.o: No such file or directory
g++: error: conv2d_1L_1CH_2x2K_bridge.o: No such file or directory
g++: fatal error: no input files
compilation terminated.
Traceback (most recent call last):

  File "/tmp/ipykernel_9905/2024476846.py", line 1, in <module>
    hls_model.compile()

  File "/home/syed/miniconda3/envs/qkeras/lib/python3.9/site-packages/hls4ml/model/hls_model.py", line 546, in compile
    self._top_function_lib = ctypes.cdll.LoadLibrary(lib_name)

  File "/home/syed/miniconda3/envs/qkeras/lib/python3.9/ctypes/__init__.py", line 460, in LoadLibrary
    return self._dlltype(name)

  File "/home/syed/miniconda3/envs/qkeras/lib/python3.9/ctypes/__init__.py", line 382, in __init__
    self._handle = _dlopen(self._name, mode)

OSError: firmware/conv2d_1L_1CH_2x2K-b9ff7a55.so: cannot open shared object file: No such file or directory