fastmachinelearning/hls4ml #186: Error during instrumenting C test bench

essa2307, 2020-03-21:
Hi,
Great work!
I have followed the example in quick start and I am getting this error
INFO: [COSIM 212-14] Instrumenting C test bench ...
   Build using "/tools/Xilinx/Vivado/2018.3/tps/lnx64/gcc-6.2.0/bin/g++"
   Compiling myproject_test.cpp_pre.cpp.tb.cpp
cosim.tv.mk:64: recipe for target 'obj/myproject_test.cpp_pre.cpp.tb.o' failed
In file included from /tools/Xilinx/Vivado/2018.3/include/floating_point_v7_0_bitacc_cmodel.h:148:0,
                 from /tools/Xilinx/Vivado/2018.3/include/hls_fpo.h:186,
                 from /tools/Xilinx/Vivado/2018.3/include/hls_half.h:44,
                 from /tools/Xilinx/Vivado/2018.3/include/etc/ap_private.h:90,
                 from /tools/Xilinx/Vivado/2018.3/include/ap_common.h:641,
                 from /tools/Xilinx/Vivado/2018.3/include/ap_int.h:54,
                 from /home/nimisha/Desktop/Esha/PSC/hls4ml/example-models/my-hls-test/firmware/parameters.h:5,
                 from /home/nimisha/Desktop/Esha/PSC/hls4ml/example-models/my-hls-test/myproject_test.cpp:25:
/tools/Xilinx/Vivado/2018.3/include/mpfr.h:181:9: error: ‘__gmp_const’ does not name a type
 typedef __gmp_const __mpfr_struct *mpfr_srcptr;
         ^~~~~~~~~~~
/tools/Xilinx/Vivado/2018.3/include/mpfr.h:181:48: error: expected constructor, destructor, or type conversion before ‘;’ token
 typedef __gmp_const __mpfr_struct *mpfr_srcptr;
                                                ^
Can you please help?

vloncar, 2020-03-21:
What distribution is this? You probably installed Vivado on an unsupported Linux platform.

essa2307, 2020-03-21:
Ubuntu 17.10.

vloncar, 2020-03-21:
From the installation [guide](https://www.xilinx.com/content/dam/xilinx/support/documentation/sw_manuals/xilinx2018_3/ug973-vivado-release-notes-install-license.pdf):

Linux Support
• Red Hat Enterprise Workstation/Server 7.2, 7.3, 7.4, and 7.5 (64-bit)
• Red Hat Enterprise Workstation 6.6, 6.7, 6.8, and 6.9 (64-bit)
• SUSE Linux Enterprise 11.4 and 12.3 (64-bit)
• CentOS 7.2, 7.3, 7.4, and 7.5 (64-bit)
• CentOS 6.7, 6.8, and 6.9 (64-bit)
• Ubuntu Linux 16.04.4, 18.04 LTS (64-bit)

You really need to update your system, Ubuntu 17.10 doesn't receive security updates anymore.

closed 2020-03-21 by vloncar