fastmachinelearning/hls4ml #187: Trouble in synthesizing MNIST model

essa2307, 2020-04-02:
Hello,
I trained my own MNIST model but synthesis is stuck. I am attaching my mnist.json and mnist.h5 files.  
For testbench, I made an x_test.dat file which contains the pixel values of 10 images from mnist. So, the file has 10 rows and 784 columns.  
y_test.dat file similarly contains 10 rols with 10 columns as the target labels.

This is my configuration file:
KerasJson: keras/mnist.json
KerasH5:   keras/mnist.h5
InputData: keras/x_test.dat
OutputPredictions: keras/y_test.dat
OutputDir: my-hls-test
ProjectName: myproject
XilinxPart: xc7a100tcsg324-1
ClockPeriod: 5
Backend: Vivado

IOType: io_parallel # options: io_serial/io_parallel
HLSConfig:
  Model:
    Precision: ap_fixed<16,6>
    ReuseFactor: 2000
    Strategy: resource
 The other lines are commented

But when I run
hls4ml build -p my-hls-test -a
The synthesis is getting stuck at this message:
***** C SIMULATION *****
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch GCC as the compiler.
   Compiling ../../../../myproject_test.cpp in debug mode
   Compiling ../../../../firmware/myproject.cpp in debug mode
   Generating csim.exe
Processing input 0
INFO: Saved inference results to file: tb_data/csim_results.log
INFO: [SIM 211-1] CSim done with 0 errors.
INFO: [SIM 211-3] *************** CSIM finish ***************
***** C SIMULATION COMPLETED IN 0h0m6s *****
***** C/RTL SYNTHESIS *****
INFO: [SCHED 204-61] Option 'relax_ii_for_timing' is enabled, will increase II to preserve clock frequency constraints.
INFO: [HLS 200-10] Analyzing design file 'firmware/myproject.cpp' ... 
WARNING: [HLS 214-114] Since the only kind of statements allowed in a dataflow region are variable declarations and function calls, the compiler may not be able to correctly handle the region: firmware/nnet_utils/nnet_dense.h:139:9
WARNING: [HLS 214-104] Only for-loops and functions support the dataflow: firmware/nnet_utils/nnet_dense.h:151:9
WARNING: [HLS 200-471] Dataflow form checks found 2 issue(s) in file firmware/myproject.cpp
INFO: [HLS 200-111] Finished Linking Time (s): cpu = 00:02:42 ; elapsed = 00:02:46 . Memory (MB): peak = 569.855 ; gain = 118.969 ; free physical = 891 ; free virtual = 5302
INFO: [HLS 200-111] Finished Checking Pragmas Time (s): cpu = 00:02:42 ; elapsed = 00:02:46 . Memory (MB): peak = 569.855 ; gain = 118.969 ; free physical = 892 ; free virtual = 5303
INFO: [HLS 200-10] Starting code transformations ...

Could you please have a look at my input files?

vloncar, 2020-04-02:
It's not stuck, it just needs a lot of time to partition the enormous weights arrays that you are trying to use. Try reducing the number of units in the Dense layers (now at 512) and applying some regularization so that the number of non-zero weights decreases.

essa2307, 2020-04-02:
I did. Now I made a small model. Now the synthesis completed but failed. Am I doing something wrong?
[issue_2.zip](https://github.com/hls-fpga-machine-learning/hls4ml/files/4420895/issue_2.zip)
Here is the error log:
[vivado_hls.log](https://github.com/hls-fpga-machine-learning/hls4ml/files/4420900/vivado_hls.log)

vloncar, 2020-04-02:
It looks like the generated project uses `PIPELINE` pragma instead of `DATAFLOW`. The issue comes from [here](https://github.com/hls-fpga-machine-learning/hls4ml/blob/master/hls4ml/writer/vivado_writer.py#L146). You can specify `Strategy: Resource` (with a capital `R`) and it will work.

essa2307, 2020-04-02:
Thanks a lot! Everything was completed.
However, I did face some issues with the test files. So I had stored values with e notation and that had a lot of errors. But when I converted to %1.4f format, all the steps were completed.
Again thanks. I am closing this issue.

closed 2020-04-02 by essa2307