fastmachinelearning/hls4ml #189: MNIST test on Vivado

essa2307, 2020-04-19:
Hi,
I have generated the hdl files and taken all the files from /myproject_prj/solution1/impl/hdl/vhdl and loaded as source files in Vivado 2018.3. Now I ran simulation by forcing a constant as the image in hex value from home/esha/myproject_prj/solution1/sim/tv/cdatafile and gave other necessary inputs:
ap_start =1
ap_rst=0
ap_clock=clock signal with 10ns prd
input1_V_ap_vld = 1
. Now I see that the output of the first layer is not working and all are 0s. And finally I am not getting an output from the last softmax layer. This is for a project that passed c/rtl cosimulation.

I actually want to implement on my FPGA, so I want to make sure the simulation is fine. Could you please help?

thesps, 2020-04-24:
How did you drive the network inputs with the sample image, is that hard coded in some HDL?

Could you check how the control signals are driven in the RTL simulation and make sure they are connected as required? Also see section Interface Synthesis Reference (page 465 onwards) of the [HLS user guide](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2019_2/ug902-vivado-high-level-synthesis.pdf#nameddest=xApplyingOptimizationDirectives).

mgrailoo, 2020-11-25:
Would you please explain the input and output range when they are in Fixed-point format?
Are you changing the range of input when you consider 6 bits for the integer part? Because for example in part1, the input range is mostly less than 4 and there is no need for a 6-bit integer part. And the output range is probably is always between zero and one. So how to quantize the output?

closed 2026-10-27 by github-actions[bot]