fastmachinelearning/hls4ml #191: Example compilation problem

AlesRuda, 2020-05-04:
Hello,

thanks for excellent work. I am using windows environment. I try compile tensorflow example using these commands

python .. \ scripts \ hls4ml.py convert -c tf-config.yml
python .. \ scripts \ hls4ml.py build -p my-hls-test -a

Simulation works, but synthesis fails on 
nested_exception.h:125:13: error: use of undeclared identifier '__ex'
    { throw __ex; }

Full output log is attached.
[out.log](https://github.com/hls-fpga-machine-learning/hls4ml/files/4577128/out.log)

Thanks in advance.

vloncar, 2020-05-04:
Windows is not supported. Try on Linux.

AlesRuda, 2020-05-04:
> Komu: hls-fpga-machine-learning/hls4ml <hls4ml@noreply.github.com>
> (#191)
> "

It is not good news :-)

Thanks

(https://github.com/hls-fpga-machine-learning/hls4ml/issues/191#issuecomment-623679243)
"

vloncar, 2020-05-04:
Unfortunately, this is a bug in Vivado HLS, and we can't do anything to fix it. There is a workaround posted [here](https://github.com/hls-fpga-machine-learning/hls4ml/pull/131#issuecomment-467811547), however the side-effects, if any, have not been explored.

closed 2020-09-23 by jmduarte