fastmachinelearning/hls4ml #192: A primitive problem

sinnnnna, 2020-05-14:
I tried to set up everything from scratch to convert a Keras model to RTL but I was not successful even in running the quick start example. 
1- I installed ubuntu 18.04 on a virtual machine
2- Downloaded and installed Vivado 19.2
3- Installed hls4ml
4- Installed following packages that I saw in the docker file: libglib2.0-0, libsm6, libxi6,libxrender1,libxrandr2,libfreetype6, libfontconfig, lsb-release, procps, gcc-multilib, g++-multilib

**Am I missing something?** Because when I execute "hls4ml build -p my-hls-test -a" from the quick start page, I face the following error:

Loading configuration from my-hls-test/hls4ml_config.yml

****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2019.2 (64-bit)
  **** SW Build 2708876 on Wed Nov  6 21:39:14 MST 2019
  **** IP Build 2700528 on Thu Nov  7 00:09:20 MST 2019
    ** Copyright 1986-2019 Xilinx, Inc. All Rights Reserved.

source /tools/Xilinx/Vivado/2019.2/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/tools/Xilinx/Vivado/2019.2/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'sina' on host 'sina-VirtualBox' (Linux_x86_64 version 4.15.0-45-generic) on Thu May 14 02:23:55 PDT 2020
INFO: [HLS 200-10] On os Ubuntu 16.04.6 LTS
INFO: [HLS 200-10] In directory '/media/sf_WorkingDirectory/hls4ml/example-models/my-hls-test'
Sourcing Tcl script 'build_prj.tcl'
INFO: [HLS 200-10] Opening project '/media/sf_WorkingDirectory/hls4ml/example-models/my-hls-test/myproject_prj'.
INFO: [HLS 200-10] Adding design file 'firmware/myproject.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'myproject_test.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'firmware/weights' to the project
INFO: [HLS 200-10] Adding test bench file 'tb_data' to the project
INFO: [HLS 200-10] Opening solution '/media/sf_WorkingDirectory/hls4ml/example-models/my-hls-test/myproject_prj/solution1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 5ns.
INFO: [HLS 200-10] Setting target device to 'xcku115-flvb2104-2-i'
INFO: [XFORM 203-101] Allowed max sub elements number after partition is 4096.
INFO: [XFORM 203-101] Allowed max sub elements number after partition is 4096.
***** C SIMULATION *****
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch GCC as the compiler.
ERROR: [SIM 211-100] CSim failed with errors.
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

aleczhanshi, 2020-08-28:
@vloncar I've come across the same issue running with the quickstart. Do you have any idea why this is the case?

Duchstf, 2020-08-28:
@aleczhanshi Hi, what model are you trying to convert? I made a quick run through the Quickstart page again and it's still working fine for me:

```
(base) [dhoang@correlator2 example-models]$ hls4ml build -p my-hls-test -a
Loading configuration from my-hls-test/hls4ml_config.yml

****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2018.2 (64-bit)
  **** SW Build 2258646 on Thu Jun 14 20:02:38 MDT 2018
  **** IP Build 2256618 on Thu Jun 14 22:10:49 MDT 2018
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

source /data/Xilinx/Vivado/Vivado/2018.2/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/data/Xilinx/Vivado/Vivado/2018.2/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'dhoang' on host 'correlator2.fnal.gov' (Linux_x86_64 version 3.10.0-957.21.3.el7.x86_64) on Thu Aug 27 22:29:48 CDT 2020
INFO: [HLS 200-10] On os "Scientific Linux release 7.8 (Nitrogen)"
INFO: [HLS 200-10] In directory '/home/dhoang/example-models/my-hls-test'
INFO: [HLS 200-10] Creating and opening project '/home/dhoang/example-models/my-hls-test/myproject_prj'.
INFO: [HLS 200-10] Adding design file 'firmware/myproject.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'myproject_test.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'firmware/weights' to the project
INFO: [HLS 200-10] Adding test bench file 'tb_data' to the project
INFO: [HLS 200-10] Creating and opening solution '/home/dhoang/example-models/my-hls-test/myproject_prj/solution1'.
INFO: [XFORM 203-101] Allowed max sub elements number after partition is 4096.
INFO: [HLS 200-10] Setting target device to 'xcku115-flvb2104-2-i'
INFO: [SYN 201-201] Setting up clock 'default' with a period of 5ns.
***** C SIMULATION *****
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch GCC as the compiler.
   Compiling ../../../../myproject_test.cpp in debug mode
   Compiling ../../../../firmware/myproject.cpp in debug mode
   Generating csim.exe
INFO: Unable to open input/predictions file, using default input.
0.0292969 0.756836 0.0546875 0.139648 0.0371094 
INFO: Saved inference results to file: tb_data/csim_results.log
INFO: [SIM 211-1] CSim done with 0 errors.
INFO: [SIM 211-3] *************** CSIM finish ***************
***** C SIMULATION COMPLETED IN 0h0m3s *****
***** C/RTL SYNTHESIS *****
INFO: [HLS 200-10] Analyzing design file 'firmware/myproject.cpp' ... 
WARNING: [HLS 214-114] Only function calls and local variable declarations are allowed in a dataflow region: firmware/nnet_utils/nnet_dense.h:139:9
INFO: [HLS 200-111] Finished Linking Time (s): cpu = 00:00:20 ; elapsed = 00:00:21 . Memory (MB): peak = 460.188 ; gain = 4.266 ; free physical = 84216 ; free virtual = 155772
INFO: [HLS 200-111] Finished Checking Pragmas Time (s): cpu = 00:00:20 ; elapsed = 00:00:21 . Memory (MB): peak = 460.188 ; gain = 4.266 ; free physical = 84216 ; free virtual = 155772
INFO: [HLS 200-10] Starting code transformations ...
I
```

aleczhanshi, 2020-08-28:
@Duchstf Thanks for the quick reply! I tried to synthesis the model with 
```
hls_model.build()
hls4ml.report.read_vivado_report('my-hls-test')
```
but I got this

```
***** C SIMULATION *****
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch GCC as the compiler.
ERROR: [SIM 211-100] CSim failed with errors.
INFO: [SIM 211-3] *************** CSIM finish ***************
4
while executing
"source build_prj.tcl"
("uplevel" body line 1)
invoked from within
"uplevel #0 [list source $arg] "
```

Briefly I just installed hls4ml and Vivado on Ubuntu 20.04. Did I miss anything?

Duchstf, 2020-08-28:
@aleczhanshi Hello, what is the vivado hls version you are using? I'm on v2018.2, and it's working for me. Also, were you trying to convert a different model rather than the standard `KERAS_3layer.json` as in the Quickstart page?

jmduarte, 2021-09-12:
Closed due to inactivity.

closed 2021-09-12 by jmduarte