fastmachinelearning/hls4ml #196: Input Features and Output Predictions File Format Documentation

Andrew-Reis-SMU-2022, 2020-06-04:
Hello,

I've had no luck in locating any documentation regarding the required format of the input and prediction files that can be included with the C simulation and C/RTL co-simulation.

For the most basic use case of 1 dimensional input and output, the expected format is clear from the cpp files generated. However, I am more concerned with the input file format for models with multiple input layers, with some of those layers consisting of multi-dimensional input (CNN image processing with multiple color channels).

This also is something I could piece together by looking through the generated code, but I wanted to see if there already existed resources that could save me the trouble.

Thanks

thesps, 2020-06-05:
This probably isn't immediately helpful, but the next release will have a method to run the simulation from the Python model object, accepting the same data type as the model. That said, I don't know if what's been developed so far would support multiple input layers, though image inputs would be fine.

vloncar, 2020-06-05:
Unfortunately, files containing multiple inputs are not supported yet. However, since this affects only the testbench and the generated HLS project supports multiple inputs, you can code your own solution in whatever way you find the most intuitive.

closed 2020-06-21 by Andrew-Reis-SMU-2022