fastmachinelearning/hls4ml #199: Biases dont get divided into parts of first layer of MNIST model

rajeshpandit107, 2020-06-16:
Compiling ../../../../firmware/myproject.cpp in debug mode
../../../../firmware/myproject.cpp:33:26: fatal error: weights/b1_0.h: No such file or directory
 #include "weights/b1_0.h"
                          ^
compilation terminated.
make: *** [obj/myproject.o] Error 1
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

Issue: Weights get divided into multiple files and biases do not. This creates file not found error. Any help upon this?

vloncar, 2020-06-17:
What branch is this? Can you provide your model, and the config file you used?

rajeshpandit107, 2020-06-17:
I am using the master branch. In keras-to-hls.py, MAXMULT is set to 4096. I am using this for model architecture of 128x128x128x10, with 784 inputs, for MNIST digit recognition.

[image]

vloncar, 2020-06-17:
Dividing weights has not been a part of the master branch for a while now. Your config file also implies old paths and old format of the config file itself (e.g., `DefaultPrecision` is not used anymore, but will probably fail for you if you remove it)

rajeshpandit107, 2020-06-17:
I tried with the latest master branch, and got the error pre-synthesis failed due to loop unrolling. Any help on this ?

ERROR: [XFORM 203-504] Stop unrolling loop 'Accum1' (firmware/nnet_utils/nnet_dense.h:184) in function 'nnet::dense_latency<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config6>' because it may cause large runtime and excessive                                     memory usage due to increase in code size. Please avoid unrolling the loop or form sub-functions for code in the loop body.
ERROR: [HLS 200-70] Pre-synthesis failed.
command 'ap_source' returned error code
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

INFO: [Common 17-206] Exiting vivado_hls at Thu Jun 18 00:17:07 2020...

Here goes the config file :

[image]

vloncar, 2020-06-17:
Put `Strategy` under `Model`, not under `HLSConfig`.

rajeshpandit107, 2020-06-17:
@vloncar   Thank you so much. That worked.

closed 2020-06-17 by rajeshpandit107