fastmachinelearning/hls4ml #200: Run into "KeyError: 'name'" when converting.

dustinjoe, 2020-06-24:
When I am trying to convert an autoencoder Keras model using command 'hls4ml convert -c keras-auto-config.yml
Loading configuration from keras-auto-config.yml'. Run into error like this:
--------------------------------------------------
Interpreting Sequential
Input shape: [200]
Topology:
Layer name: dense_1, layer type: Dense, current shape: [[None, 200]]
Layer name: activation_1, layer type: Activation, current shape: [[None, 256]]
Traceback (most recent call last):
  File "/home/xyzhou/anaconda3/bin/hls4ml", line 138, in <module>
    main()
  File "/home/xyzhou/anaconda3/bin/hls4ml", line 58, in main
    args.func(args)
  File "/home/xyzhou/anaconda3/bin/hls4ml", line 87, in convert
    model = hls4ml.converters.keras_to_hls(yamlConfig)
  File "/home/xyzhou/anaconda3/lib/python3.7/site-packages/hls4ml/converters/keras_to_hls.py", line 174, in keras_to_hls
    output_shapes[keras_layer['name']] = input_shapes #[0]
KeyError: 'name'
-----------------------------------------------
The model arch is like this:

autoencoder.summary()
Model: "sequential_1"
_________________________________________________________________
Layer (type)                 Output Shape              Param #   
=================================================================
dense_1 (Dense)              (None, 256)               51456     
_________________________________________________________________
activation_1 (Activation)    (None, 256)               0         
_________________________________________________________________
dropout_1 (Dropout)          (None, 256)               0         
_________________________________________________________________
dense_2 (Dense)              (None, 128)               32896     
_________________________________________________________________
activation_2 (Activation)    (None, 128)               0         
_________________________________________________________________
dropout_2 (Dropout)          (None, 128)               0         
_________________________________________________________________
dense_3 (Dense)              (None, 256)               33024     
_________________________________________________________________
activation_3 (Activation)    (None, 256)               0         
_________________________________________________________________
dropout_3 (Dropout)          (None, 256)               0         
_________________________________________________________________
dense_4 (Dense)              (None, 200)               51400     
=================================================================
Total params: 168,776
Trainable params: 168,776
Non-trainable params: 0
------------------------------------------------------------------
Anything I can do to deal with this?
Thank you!

vloncar, 2020-06-24:
This is because you use a Sequential model. We have fixed this issue in the branch for PR #195, so you can try using that branch (soon to be master anyway), or you can manually fix it applying the change from e492ae7b286cb5a57358ea2e4bf607157c40a610.

dustinjoe, 2020-06-24:
Thank you very much. 
This solution works.

closed 2020-06-24 by dustinjoe