fastmachinelearning/hls4ml #202: Wrong shape for concatenation layer

melko, 2020-07-02:
As far I can understand, a concatenation layer should take two tensors and append the second to the first, along the specified axis.
Eg. having T1 with shape (X, Y, Z1) and T2 with shape (X, Y, Z2), a concatenation with axis=-1 should produce a tensor with shape (X, Y, Z1+Z2).

Looking at the hls4ml code in the Concatenate class, the result shape is computed as the sum of the input shapes
`shape = [sum(x) for x in zip(inp1.shape, inp2.shape)]`
so, for the previous example, the result would be (2X, 2Y, Z1+Z2), which seems wrong to me.
The implementation code inside nnet_merge.h seems correct at a quick glance, so I guess the problem is just in the
shape calculation (resulting in instantiating arrays with a bigger size, respect to what is needed).

If you agree, I can prepare a patch to fix this issue.

vloncar, 2020-07-02:
Go for it! Thanks.

melko, 2020-07-02:
Done with PR #203 (I've included also a fix for another issue in the concatenate3d implementation).

commit c687a2b6 (2020-07-07): Fix shape calculation for the concatenate layer

The resulting shape of a concatenation should be the sum of the input
shapes only for the specified axis, leaving the other dimensions untouched.
The assumption is that the two inputs have the same shape, except for
the dimension on which the concatenation should be done.

FIX #202
closed 2020-07-07 by thesps