fastmachinelearning/hls4ml #204: CSIM Inference Output Discrepancy

veyron8800, 2020-07-11:
Hello,

I am running into an issue where HLS4ML model inference output from CSIM does not match what is produced by using tf.keras. I have compiled the model into [this repository](https://github.com/veyron8800/DigitClassification), so that the bug can be easily reproduced.

The model in question is a simplistic CNN with the following architecture:

```
model = models.Sequential()
model.add(layers.Conv2D(4, (3, 3), activation='relu', input_shape=(28, 28, 1)))
model.add(layers.MaxPooling2D((2, 2)))
model.add(layers.Conv2D(8, (3, 3), activation='relu'))
model.add(layers.MaxPooling2D((2, 2)))
model.add(layers.Conv2D(16, (3, 3), activation='relu'))
model.add(layers.Flatten())
model.add(layers.Dense(16, activation='relu'))
model.add(layers.Dense(10, activation='softmax'))
```

Unfortunately, the inference output on the test image set for the [csim of the hls4ml](https://raw.githubusercontent.com/veyron8800/DigitClassification/master/InferenceComp/HLS4ML_Out.dat) model is nowhere close to what is produced by [tf.keras](https://raw.githubusercontent.com/veyron8800/DigitClassification/master/InferenceComp/TF_OUT.dat). The thing that really caught my eye is the [hls4ml output](https://raw.githubusercontent.com/veyron8800/DigitClassification/master/InferenceComp/HLS4ML_Out.dat) is that there are values that exceed the bounds of [0,1], when the final layer has a softmax activation. This suggests to me that the problem is most likely with the [cpp definition of the softmax function](https://github.com/hls-fpga-machine-learning/hls4ml/blob/e9d057618d875f0e75015786b315d5fef63e6e84/hls4ml/templates/vivado/nnet_utils/nnet_activation.h#L174).

Here are some more links to the example repo for convenience:
Model creation and training: [ModelTraining.py](https://github.com/veyron8800/DigitClassification/blob/master/ModelTraining.py)
Config file: [DC.yml](https://github.com/veyron8800/DigitClassification/blob/master/DC.yml)
Inference output generation: [InferenceComp.py](https://github.com/veyron8800/DigitClassification/blob/master/InferenceComp.py)
Inference output: [InferenceComp](https://github.com/veyron8800/DigitClassification/tree/master/InferenceComp)

Thanks

Duchstf, 2020-07-12:
Hey, 

have you tried setting `Precision: ap_fixed<16,6>` to `Precision: float` and then compare the outputs? It might be that the weights are quantized leading to significantly different results. You can see the documentation [here](https://fastmachinelearning.org/hls4ml/setup/CONFIGURATION.html) for the bits notation. 

I've also been kind of worry about `softmax`, but let's make sure the results are similar when you use `float` before looking into it.

veyron8800, 2020-07-12:
When I make the following change in the .yml file `Precision: float` I receive the following errors in csim:

```
In file included from firmware/parameters.h:15,
                 from firmware/DC.cpp:22:
firmware/nnet_utils/nnet_pooling.h: In instantiation of ‘T nnet::pad_val() [with T = float; nnet::Pool_Op op = nnet::Max]’:
firmware/nnet_utils/nnet_pooling.h:164:90:   required from ‘void nnet::pooling2d_cl(data_T*, data_T*) [with data_T = float; CONFIG_T = config4]’
firmware/DC.cpp:74:65:   required from here
firmware/nnet_utils/nnet_pooling.h:79:11: error: request for member ‘width’ in ‘x’, which is of non-class type ‘float’
   79 |       x[x.width - 1] = 1;
      |         ~~^~~~~
In file included from firmware/parameters.h:9,
                 from firmware/DC.cpp:22:
firmware/nnet_utils/nnet_activation.h: In instantiation of ‘unsigned int nnet::softmax_idx_from_real_val(data_T) [with data_T = float; CONFIG_T = softmax_config13]’:
firmware/nnet_utils/nnet_activation.h:247:62:   required from ‘void nnet::softmax(data_T*, res_T*) [with data_T = float; res_T = float; CONFIG_T = softmax_config13]’
firmware/DC.cpp:108:82:   required from here
firmware/nnet_utils/nnet_activation.h:194:24: error: request for member ‘width’ in ‘x’, which is of non-class type ‘float’
  194 |     ap_uint<N> y = x(x.width-1, x.width-N); // slice the top N bits of input
      |                      ~~^~~~~
firmware/nnet_utils/nnet_activation.h:194:35: error: request for member ‘width’ in ‘x’, which is of non-class type ‘float’
  194 |     ap_uint<N> y = x(x.width-1, x.width-N); // slice the top N bits of input
      |                                 ~~^~~~~
firmware/nnet_utils/nnet_activation.h:194:21: error: ‘x’ cannot be used as a function
  194 |     ap_uint<N> y = x(x.width-1, x.width-N); // slice the top N bits of input
      |                    ~^~~~~~~~~~~~~~~~~~~~~~
firmware/nnet_utils/nnet_activation.h: In instantiation of ‘float nnet::softmax_real_val_from_idx(unsigned int) [with data_T = float; CONFIG_T = softmax_config13]’:
firmware/nnet_utils/nnet_activation.h:203:62:   required from ‘void nnet::init_exp_table(typename CONFIG_T::exp_table_t*) [with data_T = float; CONFIG_T = softmax_config13; typename CONFIG_T::exp_table_t = ap_fixed<18, 8>]’
firmware/nnet_utils/nnet_activation.h:234:41:   required from ‘void nnet::softmax(data_T*, res_T*) [with data_T = float; res_T = float; CONFIG_T = softmax_config13]’
firmware/DC.cpp:108:82:   required from here
firmware/nnet_utils/nnet_activation.h:186:9: error: request for member ‘width’ in ‘x’, which is of non-class type ‘float’
  186 |     x(x.width-1, x.width-N) = i;
      |       ~~^~~~~
firmware/nnet_utils/nnet_activation.h:186:20: error: request for member ‘width’ in ‘x’, which is of non-class type ‘float’
  186 |     x(x.width-1, x.width-N) = i;
      |                  ~~^~~~~
firmware/nnet_utils/nnet_activation.h:186:6: error: ‘x’ cannot be used as a function
  186 |     x(x.width-1, x.width-N) = i;
      |     ~^~~~~~~~~~~~~~~~~~~~~~
```
In the same spirit as your recommendation above, I increased the fixed precision to `ap_fixed<32,12>`. And while the output did change substantially, it appears to have only shuffled the nonsensical activations. 
[HLS4ML_Out ap_fixed-32-12.txt](https://github.com/hls-fpga-machine-learning/hls4ml/files/4907828/HLS4ML_Out.ap_fixed-32-12.txt)

thesps, 2020-07-12:
Hi, and thanks for providing the example, that was very useful. Perhaps you could see what results you get with this as the HLSConfig in DC.yml:
```
 HLSConfig:
   Model:
     Precision: ap_fixed<16,6>
     ReuseFactor: 60000
     Strategy: Resource
   LayerName:
     dense_1_softmax:
       exp_table_t: ap_fixed<18,8,AP_RND,AP_SAT>
       inv_table_t: ap_fixed<18,8,AP_RND,AP_SAT>
```
I think this improves things, although it's still not perfect.
Basically, the MNIST model predicts one class with much greater probability than the other 9, and this can create rounding/saturation issues in the softmax. The above changes will bound the outputs between 0 & 1, but there may still be some cases where an output which should be close to 0 is actually computed as close to 1. That ought to be something we can fix.

If you're able, would you be able to repeat the test with the previous hls4ml tag? If it's annoying to get working because of the API changes, then don't worry. I would like to isolate whether the issue exists in that version also.

veyron8800, 2020-07-12:
Hello @thesps ,

Thanks for looking into this for me. Yes, the output now after making that type adjustment is looking much better. I can also see those cases where activation values very near zero are creeping closer to 1/2 or even 1. If you can get that sorted, then I think everything should be good.

By previous hls4ml tag, I assume you are referring to v0.1.6? If so, then I do have some interesting results. It looks like the softmax definition for that version does not suffer from this same issues. Please see the attached. It was generated with just the precision defined as `ap_fixed<16,6>`.
[csim_results_0_1_6.log](https://github.com/hls-fpga-machine-learning/hls4ml/files/4907938/csim_results_0_1_6.log)

thesps, 2020-07-13:
Are those predictions correct for your model? Perhaps you could report the model accuracy, ideally for the Keras model as well as hls4ml with both Softmax versions.

I made a new Softmax implementation, intended to be more numerically stable than the others. You can find it [here](https://github.com/thesps/hls4ml/tree/stable_softmax). You might like to try it out also.
I'm not sure yet if I want to replace the existing version with it, since it's very slightly slower and more resource intensive than the others. If you'd like to try it, let me know how you get on. I also just hard coded one thing that would need to be replaced for a production version.

You'd need to be a bit careful setting the data types again. I think the following should be okay:
``` 
HLSConfig:
   Model:
     Precision: ap_fixed<16,6>
     ReuseFactor: 60000
     Strategy: Resource
   LayerName:
     dense_1:
       result: ap_fixed<16,6,AP_RND,AP_SAT>
     dense_1_softmax:
       exp_table_t: ap_fixed<18,3,AP_RND,AP_SAT>
       inv_table_t: ap_fixed<18,3,AP_RND,AP_SAT>
```

veyron8800, 2020-07-17:
Hello @thesps ,

I got a chance to run some accuracy tests on the different branches, and it looks like the new numerically stable softmax implementation you linked above gets the job done wonderfully.

When run through tensorflow, the model has a categorical accuracy of 97.16%. HLS4ML when run on the master branch with the type adjustments in the config file mentioned above produces an accuracy of 81.82%. The stable softmax branch linked above matches the 97.16% accuracy with or without the type adjustments in the config file.

Thank you for the quick turnaround on this issue!

closed 2020-09-14 by veyron8800