fastmachinelearning/hls4ml #216: A number of the example models do not work

whiskid, 2020-08-23:
I am attempting to use the tool and all the example models containing conv2d layers that I have tried fail with the error:

INFO: [HLS 200-489] Unrolling loop 'ConvOutHeight' (firmware/nnet_utils/nnet_conv2d.h:245) in function 'nnet::conv_2d_latency_cl<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config6>' completely with a factor of 13.
ERROR: [XFORM 203-504] Stop unrolling loop 'ConvOutHeight' (firmware/nnet_utils/nnet_conv2d.h:245) in function 'nnet::conv_2d_latency_cl<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, config6>' because it may cause large runtime and excessive memory usage due to increase in code size. Please avoid unrolling the loop or form sub-functions for code in the loop body.
ERROR: [HLS 200-70] Pre-synthesis failed.
command 'ap_source' returned error code
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

Since they're example models, I assume they work on SOME devices and with SOME versions of Vivado at least--what machines and versions are those? I am trying to use the tool to implement a small keras model with a conv2d layer, and am not sure this tool is right for it.

whiskid, 2020-08-24:
note: applies to all the example models containing conv1d layers as well. i am running vivado 2020.1 on Ubuntu 18.04

jmduarte, 2020-09-23:
We've updated the example models repository recently. Maybe @Duchstf can comment on which ones are expected to work.

Duchstf, 2020-09-24:
Hello, 

You can check out [this repo](https://github.com/hls-fpga-machine-learning/example-models) for all the example models. After that you can use our API to load and convert the models.

For CNN in particular, the small ones should work (e.g `KERAS_conv1d_small.json`). However, other models like `KERAS_conv1d.json` or `KERAS_conv2d_model.json` often run into synthesis problems in newer vivado hls versions. You might try to synthesize those using vivado 2017.2 (at least it worked for some people).

There are currently efforts to convert bigger CNN models using streaming implementations, for example:

1.  PR #220 

2. https://github.com/violatingcp/hls4ml/tree/sc20_test

Keep in mind that they are still experiment branches, so everything will not work smoothly. But at the very least you should be able to look at the hls templates to see how the bigger convolutional layers are implemented. I believe there are also some example projects in them.

simplelins, 2020-12-04:
Hi, @Duchstf  
      for the branch https://github.com/violatingcp/hls4ml/tree/sc20_test,  which model that contains convolution such as resnet50 I can run with vivado 2020,  thanks

Duchstf, 2020-12-04:
@simplelins There are a few Resnet-50 models here, can you try them? https://github.com/violatingcp/hls4ml/tree/sc20_test/example-models

simplelins, 2020-12-07:
ok, I will try to find a example that can run with vivado 2020 and vitis 2020.

simplelins, 2020-12-07:
models | vivado 2020.1 | vitis   2020.1
-- | -- | --
deepcalo | pass | pass
resnet50-hls-full_v2 | pass | generate solution：    ERROR: [HLS 200-642] The   'config_array_partition -maximum_size' command is not supported.   C/RTL SYNTHESIS：   ERROR:   [HLS 214-126] An Array of user defined type 'gpu_0_data_0' in function   'myproject_layer' cannot be used  as an   axi slave stream or as a fifo. Use hls::stream<'gpu_0_data_0'> instead:   /home/lww/project/12_FPGA/sc20_test_hls4ml/example-vitis/resnet50-hls-full_v2/firmware/myproject_layer.cpp:30   ERROR:   [HLS 214-126] An Array of user defined type 'layer5_out' in function   'myproject_layer' cannot be used  as an   axi slave stream or as a fifo. Use hls::stream<'layer5_out'> instead:   /home/lww/project/12_FPGA/sc20_test_hls4ml/example-vitis/resnet50-hls-full_v2/firmware/myproject_layer.cpp:31
resnet50-hls-full_300mus | C/RTL SYNTHESIS：   firmware/myproject.h:32:5:   error: invalid storage class specifier in function declarator   firmware/myproject.cpp:301:30:   error: no matching function for call to 'add'   3   warnings and 19 errors generated | C/RTL SYNTHESIS：   ERROR:   [HLS 214-126] An Array of user defined type 'gpu_0_data_0' in function   'myproject' cannot be used  as an axi   slave stream or as a fifo. Use hls::stream<'gpu_0_data_0'> instead:   /home/lww/project/12_FPGA/sc20_test_hls4ml/example-vitis/resnet50-hls-full_300mus/firmware/myproject.cpp:134
resnet50-hls-full | build_prj_full.tcl：   firmware/myproject_full.cpp:227:77:   error: use of overloaded operator '<<' is ambiguous (with operand types   'basic_ostream<char, std::char_traits<char> >' and 'input_t' (aka   'ap_fixed<8, 6>'))   build_prj_1stbit.tcl：   firmware/myproject_1stbit.cpp:106:1:   error: use of undeclared identifier 'c'   build_prj_2ndbit.tcl：   firmware/myproject_2ndbit.cpp:162:23:   error: no matching function for call to 'split'   build_prj_3rdbit.tcl：   firmware/myproject_3rdbit.cpp:181:24:   error: no matching function for call to 'split'   build_prj_4thbit_stream.tcl：   ERROR: [HLS   200-70] Compilation errors found: In file included from   firmware/myproject_4thbit_stream.cpp:1:   firmware/myproject_4thbit_stream.cpp:121:23:   error: no matching function for call to 'split'    if(!input[0].empty())nnet::split<layer150_t,layer150_t,config154>(input,layer150_out1,layer150_out2);                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~   firmware/nnet_utils/nnet_merge.h:78:6:   note: candidate template ignored: substitution failure [with input_T =   ap_fixed<16, 6, 5, 3, 0>, res_T = ap_fixed<16, 6, 5, 3, 0>,   CONFIG_T = config154]       build_prj_4thbit.tcl：   firmware/myproject_4thbit.cpp:134:23:   error: no matching function for call to 'split'    if(!input[0].empty())nnet::split<layer150_t,layer150_t,config154>(input,layer150_out1,layer150_out2);                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~   firmware/nnet_utils/nnet_merge.h:78:6:   note: candidate template ignored: substitution failure [with input_T =   ap_fixed<16, 6, 5, 3, 0>, res_T = ap_fixed<16, 6, 5, 3, 0>,   CONFIG_T = config154] |  
conv1d_test | pass | ERROR:   [XFORM 203-133] Cannot reshape array 'layer_in_row.Array.V.1' : it is used in   memory shift read access.
dc_test_3_exp7 | C/RTL SYNTHESIS：   ERROR: [XFORM   203-711] Internal global variable 'layer_in.V'  failed dataflow checking: it can only be   read in one process function.   WARNING: [XFORM   203-713] Internal global variable 'layer_in.V' has read and write operations   in process function 'nnet::cnnshift<ap_fixed<16, 6, (ap_q_mode)5,   (ap_o_mode)3, 0>, ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>,   config2>'. |  
shomodel | C/RTL SYNTHESIS：   firmware/parameters.h:1:2:   error: unterminated conditional directive   #ifndef   PARAMETERS_H_    ^   1 error   generated |

Duchstf, 2020-12-07:
@violatingcp FYI

simplelins, 2020-12-08:
@Duchstf   hi， how to generate source code for resnet50-hls-full_v2 by myself ? I try to generate the code using the config file  "https://github.com/violatingcp/hls4ml/blob/sc20_test/example-models/resnet50-hls-full_v2/hls4ml_config.yml" , but so much difference,  what's the right way?  
 command as below: 
hls4ml convert -c hls4ml_config.yml
and the tool "hls4ml" is also from sc20_test

closed 2023-04-09 by jmduarte