fastmachinelearning/hls4ml #224: Configuration generator makes no entry for "Input" Layer

thaarres, 2020-10-15:
I am trying to set the precision of each layer type through the nested config dictionaries in the following way:
```
hls_cfg = hls4ml.utils.config_from_keras_model(m,granularity='type')
hls_cfg['Model'] = {}
...
hls_cfg['Model']['Precision'] = 'ap_fixed<8,4>'
hls_cfg['LayerType']['MaxPooling2D']['Precision']  = 'ap_fixed<8,6>'
...
```
The above works fine, however, when I try to set the input layer precision
```
hls_cfg['LayerType']['Input']['Precision']  = 'ap_fixed<16,6>'
```
I get
```
File "benchmarkModels.py", line 34, in toHLS
    hls_cfg['LayerType']['Input']['Precision'] = 'ap_fixed<16,6>'
KeyError: 'Input'
```
It seems the input layer is not generated by the configuration function, and to make it work I need
`hls_cfg['LayerType']['Input'] = {'Precision':'ap_fixed<16,6>'}`
It would be great to have this fixed in the future so that this key is generated as well!

husnainmubarik, 2020-10-24:
Whenever I change the granularity of config from "model" to "name". My config generation breaks. Anyone have this issue?

[image]

vloncar, 2020-10-24:
Can you please try the master branch? Both issues you reported are solved in master branch. Also, please open new issues if you don't experience the exact problem that is reported.

closed 2026-10-27 by github-actions[bot]